jlas1/Klicky-Probe #8: Probe_Calibrate Error

dsmith777, 2021-07-22:
Hello,

When running probe_calibrate with the Klicky probe, I get an error when I try to accept the results.

Error:
Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT.

If I modify Probe_Calibrate in the klicky-probe.cfg, and comment out a few lines (Dock_Probe and Restore GCODE), I am able to manually remove the probe and finish the probe calibration with a successful accepting of the results.

Lines commented out:
[image]

This is my first time creating an issue on GitHub, so if you need anything further from me, just ask.

Thanks,

Dave

dsmith777, 2021-07-22:
After some discourse on the Voron Discord server, alexz V2.660 posted his workaround/fix for the issue.
https://github.com/zellneralex/klipper_config/blob/master/magprobe.cfg#L202
I will modify/add his macros that deal with this issue to the klick-probe.cfg, and get back to you guys.

dsmith777, 2021-08-04:
I tried adding those macro's from Alexz, but it didn't work for me.

jlas1, 2021-08-04:
Hi, so commenting these two lines did not solve the issue?

[image]

dsmith777, 2021-08-05:
Commenting those lines out fixed the issue, but I had to remove the probe manually to finish the probe_calibrate.

dukeduck1984, 2021-08-10:
Hi, I have the same issue.  Is there a fix rather than the manual workaround mentioned by @dsmith777 ?

QnzLAMBONE, 2021-08-16:
This should work. Relocated the save_gcode_state to after the probe_calibrate is performed

[image]

jlas1, 2021-08-16:
Will try this, thanks

dsmith777, 2021-08-17:
QnzLAMBONE's change, worked for me.

QnzLAMBONE, 2021-08-17:
It should actually look like this so that you restore the location from before attaching the probe, and then you restore the location of where the bed is probed.

[image]

Xperie, 2021-08-20:
I tried all the suggested solutions and I still get "Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT." Can I manually add the value it gives me to printer.cfg?

QnzLAMBONE, 2021-08-20:
What you may have to do is manually tell your toolhead to move to the center of your bed, like your homing override would do. What happens is if you perform a G28 and then try to probe_calibrate, during the G28 it automatically docks the probe as part of the homing sequence. So probe_calibrate sees the docking location as the original location prior to running the command. So something like "G1 X125 Y125 Z5" before "probe_calibrate" will allow you to probe the center of the bed. And with the changes mentioned above the calibration should work after that

QnzLAMBONE, 2021-08-20:
What could also be done is you change the attach_probe macro so that it sends the toolhead to the center of the bed. And you can then comment out the first save_gcode_state and restore_gcode_state lines.

Xperie, 2021-08-20:
I tried with both the original Macro with the two lines commented out and with your updated one. Home my toolhead, then I run ATTACH_PROBE and then do G1 X175 Y175 Z10. I then run the PROBE_CALIBRATE command and it does its thing. After which it asks me to input ACCEPT or ABORT. I've tried both leaving the probe on and inputting ACCEPT and taking the probe off and inputting ACCEPT both give me the error. But with your updated code it gives about a 30 second pause before giving me the  "Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT." error if I leave the probe on. Every other time it immediately gives me an error.

QnzLAMBONE, 2021-08-20:
Can you post your config for klicky_probe.cfg, I could take a look and see if I spot anything

Xperie, 2021-08-20:
`# This macro was provided by discord user Garrettwp to whom i give my thanks for sharing it with me.
# I have tweaked it a lot.
#
# this macro is  based on the great Annex magprobe dockable probe macros "#Originally developed by Mental, modified for better use on K-series printers by RyanG and Trails"
# that macro can be found here https://github.com/Annex-Engineering/Annex-Engineering_Other_Printer_Mods/blob/master/All_Printers/Microswitch_Probe/Klipper_Macros/dockable_probe_macros.cfg
#
# by standing on the shoulders of giants, lets see if we can see further

[probe]
pin: PG11 # input pin for the mag-probe
x_offset: 0
y_offset: 19.75
z_offset: 6.42
speed: 4.0
lift_speed: 8.0
samples: 5
samples_result: median
sample_retract_dist: 1.5
samples_tolerance: 0.01 #0.007
samples_tolerance_retries: 10

[gcode_macro User_Variables]
variable_verbose:               True  # Enable verbose output
variable_travel_speed:          200   # how fast all other travel moves will be performed when running these macros
variable_dock_speed:            50    # how fast should the toolhead move when docking the probe for the final movement
variable_release_speed:         100    # how fast should the toolhead move to release the hold of the magnets after docking
variable_z_drop_speed:          20    # how fast the z will lower when moving to the z location to clear the probe
variable_home_z_height:         15    # Z when homing

# if a separate Z endstop switch is in
# use, specify the coordinates of the switch here (Voron).
# Set to 0 to have the probe move to center of bed
variable_z_endstop_x:           232
variable_z_endstop_y:           347
#variable_z_endstop_x:          0
#variable_z_endstop_y:          0

# location to park the toolhead
variable_park_toolhead:         False  # Enable toolhead parking
variable_parkposition_x:        175
variable_parkposition_y:        175
variable_parkposition_z:        30

#dock location
variable_docklocation_x:        45    # X Dock position
variable_docklocation_y:        350   # Y Dock position
variable_docklocation_z:        15    # Z dock position
variable_dockarmslenght:        30    # Dock arms lenght, toolhead movement necessary to clear the dock arms

#Umbilical to help untangle the umbilical in difficult situations
variable_umbilical_x:           15    #X umbilical position
variable_umbilical_y:           15    #Y umbilical position
variable_umbilical:             False #should we untabgle the umbilical

# Do not modify below
gcode:
    {% set Mx = printer['configfile'].config["stepper_x"]["position_max"]|float %}
    {% set My = printer['configfile'].config["stepper_y"]["position_max"]|float %}
    {% set Ox = printer['configfile'].config["probe"]["x_offset"]|float %}
    {% set Oy = printer['configfile'].config["probe"]["y_offset"]|float %}
    {% set Oz = printer['configfile'].config["probe"]["z_offset"]|float %}

    # if docklocation_z is zero, use Home Z height for safety
    {% if docklocation_z == 0 %}
        SET_GCODE_VARIABLE MACRO=Probe_Variables VARIABLE=docklocation_z VALUE={ home_z_height }
    {% endif %}

    # If x, y coordinates are set for z endstop, assign them
    {% if z_endstop_x != 0 or z_endstop_y != 0 %}
        SET_GCODE_VARIABLE MACRO=Probe_Variables VARIABLE=z_endstop_x VALUE={ z_endstop_x }
[... 502 lines cut ...]

    # if not attached
    {% if P and params.ACTION == 'attach' %}
        { action_raise_error("Probe attach failed!") }
    {% endif %}

# Park Toolhead Routine
[gcode_macro Park_Toolhead]
gcode:
    {% set P = printer["gcode_macro User_Variables"].park_toolhead %}
    {% set Px = printer["gcode_macro User_Variables"].parkposition_x %}
    {% set Py = printer["gcode_macro User_Variables"].parkposition_y %}
    {% set Pz = printer["gcode_macro User_Variables"].parkposition_z %}
    {% set St = printer["gcode_macro User_Variables"].travel_speed * 60 %}
    {% set V = printer["gcode_macro User_Variables"].verbose %}

    {% if (P and 'xyz' in printer.toolhead.homed_axes) %}
        {% if V %}
            { action_respond_info("Parking Toolhead") }
        {% endif %}
        G90
        G1 X{Px} Y{Py} Z{Pz} F{St}
    {% endif %}`

This is my klicky_probe.cfg. Thanks for trying to help.

QnzLAMBONE, 2021-08-21:
> https://pastebin.com/nBPHXTK0
> 
> This is my klicky_probe.cfg. Thanks for trying to help.

In the config you sent you have the first RESTORE_GCODE_STATE commented out. The one right after Attach_Probe, without restoring the location after attaching the probe you'd end up probing at your dock instead

Xperie, 2021-08-21:
Ok I did the changes you said and I noticed something weird. The Toolhead drifts off to Y 194.75 after probing the bed. I have a feeling thats whats causing the issue but I have no idea why its doing it.

Xperie, 2021-08-21:
So what its doing when I PROBE_CALIBRATE now after homing is it grabs the probe > parks in the center at 175,175,15 > Begins probing > after probing it drifts over to Y 194.75 > saves that location > docks the probe > parks > then restores the 175,194.75,15 location.

Xperie, 2021-08-21:
Well I figured out the drifting its moving the toolhead to the probes offset afterwards. Is that supposed to happen?

Xperie, 2021-08-21:
I figured it out thanks for the help

QnzLAMBONE, 2021-08-21:
> Well I figured out the drifting its moving the toolhead to the probes offset afterwards. Is that supposed to happen?

Yes that is supposed to happen, thats how you set your offset after you probe the bed, it moves the nozzle to right above where you probed. Glad we got it figured out, what was wrong?

Xperie, 2021-08-21:
Not really sure, I reset my printer.cfg to default and then went through and set everything up again and it just worked.

jlas1, 2021-08-24:
I have pushed an updated PROBE_CALIBATE macro based of the feedback from this issue, it seems to work, it is working for me.
I'll close the issue, if necessary we will reopen it.

closed 2021-08-24 by jlas1

lrstanley, 2021-08-27:
I think there are a few issues caused by the recent changes in https://github.com/jlas1/Klicky-Probe/commit/0eb982421553e31b5fccbc38a7efbbcf2708da0f, and maybe this is also me being new to the macros, but...

PROBE_ACCURACY will error because the first gcode save starts with a prefix, but then it tries to restore without a prefix:

https://github.com/jlas1/Klicky-Probe/blob/670bf00428b20fde11c5511d73d1c58567a0776f/Klipper%20macro/klicky-probe.cfg#L389-L407

Specifically `_original_nozzle_location` vs `original_nozzle_location`

Secondly, QGL, and similar commands work without issue for me, but with the macro that's currently in this repo, if I:

1. correct the above issues
2. probe X, Y, and Z (not using klicky as a z-endstop)
3. run PROBE_ACCURACY or PROBE_CALIBRATE

It will:

1. Drop or raise to necessary configured Z homing height
2. Attach probe
3. Move to Z-Endstop position
4. Lower as if it's going to use the z-endstop, but crash right into it

Again, QGL, and similar work without issues, so I suspect it's specifically with PROBE_ACCURACY and PROBE_CALIBRATE, I just commented out some of the save/restore gcodes, as follows:

```
    # Go to Z safe distance before saving location in order to
    # avoid crashing the probe on the bed when coming back
    G1 Z{Hzh} F{Sz}
    M400 # mandatory to save the new safe position
    SAVE_GCODE_STATE NAME=_original_nozzle_location

    CheckProbe action=query
    Attach_Probe

    #Restore nozzle location to probe the right place
    #RESTORE_GCODE_STATE NAME=_original_nozzle_location MOVE=1 MOVE_SPEED={St}

    _PROBE_ACCURACY {% for p in params
            %}{'%s=%s ' % (p, params[p])}{%
           endfor %}

    #store current nozzle location
    # SAVE_GCODE_STATE NAME=_original_nozzle_location 
	
	Dock_Probe
	
    # Restore nozzle location again at the end
    RESTORE_GCODE_STATE NAME=_original_nozzle_location MOVE=1 MOVE_SPEED={St}
```

The above (which saves before attaching, and after docking) simply moved it right onto the bed and probed about 10mm away from the dock location, and ran the accuracy test like expected. It didn't try and do anything with the Z-Endstop.

I'm not sure if the above still solves whatever the original problems were, but it prevented the nozzle crash for me...