jlas1/Klicky-Probe #13: Is Omron D2F-01L good as well?

phajduk, 2021-08-22:
Hey. In Voron 2.4 sourcing there's listed [Omron D2F-01L](https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/D2F-01L/83264) switch vs suggested [Omron D2F-5](https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/D2F-5/8593136). Is it possible to use same switch with this mod? I can prepare a small PR with docs update if that's possible.

TravisWilder, 2021-08-23:
I used the D2F-01L as I was not able to source the D2F-5
If you have any sourcing link for germany your are welcome :)
I even did not found D2F-5 on Ali

jlas1, 2021-08-23:
You can also use the D2F-01 with or without the L , it should work fine, the D2F-5 is more used and better known, there is anecdotal evidence that it it better, but that is not quantified in relation to temperature variance and endurance.

closed 2021-08-23 by jlas1