jlas1/Klicky-Probe #25: Found some issues with the macro caused by nesting macros

dancat, 2021-11-24:
Hi,

I have a corexy printer with moving bed (tiny-M) and I am using the macros from Klicky probe. I found a problem that during the homing override procedure, the bed is moved to the safe distance even if this is not needed. This is not good if the bed is at its lowest when the printer tries to move it.
While investigating this, I found the following in the klipper docs (https://www.klipper3d.org/Command_Templates.html): Important! Macros are first evaluated in entirety and only then are the resulting commands executed. If a macro issues a command that alters the state of the printer, the results of that state change will not be visible during the evaluation of the macro. This can also result in subtle behavior when a macro generates commands that call other macros, as the called macro is evaluated when it is invoked (which is after the entire evaluation of the calling macro).

The problem here is that when the AttachProbe macro is called on its own, the checks to move or not the bed work because checking if Z is homed will give the real answer. If the AttachProbe macro is called from anothe macro (like HomingOverrirde) the check if Z is homed will no longer give the real status, but will give it as if the calling macro was already executed (so it will always say Z is homed).

I guess this could be avoided by using some global variables instead of the home axis check in the macros themselves. I will have a go to this and see if I can make it work. But I thought to let you know in case this is something you were not aware of. This thing might affect all the checks done to the printer status from within macros if they are called nested, so probably the safest is to always check global variables in macros...

Best regards,
Daniel

jlas1, 2021-11-24:
Hi Dancat, can you check the latest macros, it should have that issue solved if the bed is configured in the gravity variable.
It will not move the bed anymore before homing.

jlas1, 2021-11-24:
with what probe are you using the klycky macros, just curious.

dancat, 2021-11-24:
Hi, thanks for feedback. I used an euclid probe. I'll have a look again at the last version and get back with feedback.

Best regards,
Daniel

closed 2021-11-27 by jlas1