jlas1/Klicky-Probe #27: D2F-01F tolerance problem

BrownRhined, 2021-12-13:
I purchased D2F-01F
Cold test error 0.02
But thermal test

04:16:29 Too many retries
04:16:29 Retries: 5/5 Probed points range: 0.025000 tolerance: 0.010000
04:16:29 Making the following Z adjustments:
stepper_z = -0.020592
stepper_z1 = -0.008719
stepper_z2 = -0.003670
stepper_z3 = 0.032981

04:15:56 Retries: 4/5 Probed points range: 0.047500 tolerance: 0.010000
04:15:56 Making the following Z adjustments:
stepper_z = -0.036870
stepper_z1 = 0.003566
stepper_z2 = 0.049001
stepper_z3 = -0.015697

04:15:33 Retries: 3/5 Probed points range: 0.050000 tolerance: 0.010000
04:15:33 Making the following Z adjustments:
stepper_z = -0.003463
stepper_z1 = 0.065622
stepper_z2 = -0.077409
stepper_z3 = 0.015250

04:15:06 Retries: 2/5 Probed points range: 0.075000 tolerance: 0.010000
04:15:06 Making the following Z adjustments:
stepper_z = 0.040161
stepper_z1 = -0.121584
stepper_z2 = 0.109453
stepper_z3 = -0.028030

04:14:43 Retries: 1/5 Probed points range: 0.067500 tolerance: 0.010000
04:14:43 Making the following Z adjustments:
stepper_z = -0.007756
stepper_z1 = 0.103486
stepper_z2 = -0.073116
stepper_z3 = -0.022614

Should I replace the switch?
Or change the settings?

jlas1, 2021-12-13:
Can you do a probe_accuracy when hot?
Also, make sure the back magnets are touching, it improves the stability of the probe.

BrownRhined, 2021-12-14:
probe_accuracy when hot

probe accuracy results: maximum 6.437500, minimum 6.042500, range 0.395000, average 6.125500, median 6.051250, standard deviation 0.151141

probe accuracy results: maximum 6.665000, minimum 6.562500, range 0.102500, average 6.612000, median 6.608750, standard deviation 0.032265

probe accuracy results: maximum 6.437500, minimum 6.042500, range 0.395000, average 6.125500, median 6.051250, standard deviation 0.151141

jlas1, 2021-12-14:
Please post the complete probe_acccuracy, but try with another switch, sometimes one i much better than the other.

closed 2022-01-02 by jlas1