jlas1/Klicky-Probe #37: Voron 2.4, fluid Pi Spider board

stampcollector, 2022-01-02:
getting this error when i include the klicky config files 

homing_override and safe_z_homing cannot be used simultaneously

I want to retain the stock z endstop and use it to automatically set the z offset to the nozzle so i included the klinky-config and z calibration.cfg

jlas1, 2022-01-02:
That's expected, you need to comment the safe_z_homing to use the klicky macros.
You can still use the Z endstop, by editing the variables on the beginning of klicky-probe.cfg to indicate to the macro where is your Z endstop located.

stampcollector, 2022-01-02:
Ok, Thank you. I assumed that was the case but I wanted to make sure.

closed 2022-01-05 by jlas1

CD1010, 2023-05-08:
Ok, where is this safe_z_homing that i am supposed to comment out?  
I tried commenting save_z_home in printer.cfg and got different error