jlas1/Klicky-Probe #39: Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT.

pforrmi, 2022-01-07:
Hi,

I found the problem at #8. But as I understand it should be fixed with the new klicky-probe.cfg script which changed 7 days ago. But in my case if I try to do the Probe Calibration and want to accept I get the same error: Manual probe failed! Use TESTZ commands to position the nozzle prior to running ACCEPT.

Any suggestions? What else can I try?

jlas1, 2022-01-07:
Try doing it by hand, do:
ATTACH_PROBE
G0 "move to the middle of the bed"
_PROBE_CALIBRATE

now remove the probe by and and complete the procedure.

jlas1, 2022-01-08:
you can now test the new macros, this problem should be solved.

pforrmi, 2022-01-08:
Seems to be worked great now with the new macros!

It would be a dream if in future the probe does not have to be removed manually, but anyway: GREAT WORK! Thank you very much for the quick fix and the klicky probe itself.

closed 2022-01-08 by pforrmi