jlas1/Klicky-Probe #46: Dock move documentation

KlausHans, 2022-01-13:
Hello, 

maybe i am to dumb, but i got a problem setting this up. In klicky-variables.cfg there is this section
```
#Check the documentation
#Dock move 
Variable_dockmove_x:          1000    # Final toolhead movement to release
Variable_dockmove_y:             0    # the probe on the dock
Variable_dockmove_z:             0    # (can be negative)
#Attach move 
Variable_attachmove_x:        1000    # Final toolhead movement to Dock
Variable_attachmove_y:           0    # the probe on the dock
Variable_attachmove_z:           0    # (can be negative)
```
I am lost here. Nether does i know, what these parameter mean (whats is exactly "final movement to release"?), nor can i find a documentation to "Check the documentation". 
What do i need to put in here?
Currently i got a `Move out of range: -848.000  [...]` and i think these parameters are the root cause. Correct?

KlausHans, 2022-01-13:
Thank you very much for the quick reply and for the link. I oversaw this. The defaults of 1000 both for x were very confusing for me. With the provided defaults in the link it works fine and now i understand what these parameter mean. :)

closed 2022-01-13 by KlausHans

jlas1, 2022-01-13:
To be fair, the defaults of 1000 is precisely to cause a move error, there are many printers supported now that i feel it's dangerous to keep anything else as a default.