jlas1/Klicky-Probe #47: automatically detect bed size

jlas1, 2022-01-16:
Need to verify the possibility of detecting the bed size, using  Z tilt, gantry level, mesh or screws position

TravisWilder, 2022-02-18:
would this help ? (klipper config macro)
{% set min = printer.toolhead.axis_minimum %}
{% set max = printer.toolhead.axis_maximum %}

jlas1, 2022-02-24:
the toolhead min and max would not help, going outside those bounds is already protected, however, reading the link you pasted, this might work:
printer.bed_mesh.mesh_max
printer.bed_mesh.mesh_min

Now i only need to be able to work with the tuple that is output from those functions.

jlas1, 2022-02-24:
i believe i found a way, not very elegant, but still, seems to work {% set (minx,miny) = printer.bed_mesh.mesh_min %}