jlas1/Klicky-Probe #49: Microswitches

erazor666, 2022-01-20:
Hello, I want to build a probe as part of an update of av 2.4 voron. However I’m having problems locating the mentioned microswitches on AliExpress. There are a multitude of them, just not the ones mentioned. Can anyone provide a link/name of the switches I can order?

Also, is this setup ready/being made ready for the new stealtburner toolhead?

TravisWilder, 2022-01-20:
I use these https://s.click.aliexpress.com/e/_9fniiG

TravisWilder, 2022-01-20:
D2F-5L - like in the BOM is what I ordered there

jlas1, 2022-01-21:
I have used the D2F-L from https://s.click.aliexpress.com/e/_ACQoLO, they are accurate and come in packs of 10, so you can have spares.
Klicky is already compatible with Stealthburner, it fits on the same place as the Afterburner.
The magnets i purchased from here https://s.click.aliexpress.com/e/_97e2rO.

closed 2022-01-21 by jlas1