jlas1/Klicky-Probe #51: homing_override wont do save z

Jaaannis, 2022-01-23:
Hi,
when doing a G28 after reboot, Z do not move for a save XY homing procedure.
it seems there is an issue, where homing_override section check if it is necessary to move z for that.
Thanks for any help :-)

jlas1, 2022-02-02:
Sorry for the lack of reply, i was without internet access, can you give me the Gcode command sequence that you use, what the printer does and what was your expectation?

Jaaannis, 2022-02-03:
Sounds like a great time ;-)

After reboot firmware (Klipper do not know where the printhead is) with all macros included, i run a G28. 
The printer than homes X than Y after that Z.
I expected (i think it is standart in klipper for safe homing) that when I run a G28, Z Lift a bit, after that X , Y, Z home.

When as an example the klicky probe is already docked and very close to the bed it might be crash.

jlas1, 2022-02-10:
you have a configuration to enable that behavior, variable_enable_z_hop, if set to true will move Z "up" before homing.

Jaaannis, 2022-02-12:
Thank you ! 
From discription i never came to this.
Have a nice day

closed 2022-02-12 by Jaaannis