jlas1/Klicky-Probe #59: Magnets magnetizes the nozzle

bjoern79de, 2022-02-17:
I'm running Voron2.4/afterburner with Dragon-Hotend.
When the toolhead travels over the probe, the heatblock/nozzle gets magnetized, so that the nozzle pulls the endstop rod out.
I've solved it by keeping the retract low and adding an additional "horizontal retract" afterwards.

jlas1, 2022-02-17:
I think you will find that what is happening is that the endstop rod is being pulled by the AB mount magnets.
What I see pulling the endstop sometimes is a little bit of melted plastic on the tip, than when squashed functions for a while as a glue.

bjoern79de, 2022-02-18:
I can preclude melted plastic, because I removed the filament and used a new clean (and cold) nozzle.
When I travel directly to the endstop, the endstop rod isn't pulled by the AB mount magnets. But when the AB-mount travels over the probe (a few seconds ago) the endstop-rod keeps hanging under the nozzle tip.

jlas1, 2022-02-18:
can you send me an image of a movie, never heard of that before.

TravisWilder, 2022-02-18:
the nozzle should never touch an come close to the magnets - so this seem very strange

jlas1, 2022-03-02:
Could you post the images so that we can troubleshoot this further?

bjoern79de, 2022-03-03:
Sorry, I've replaced the hotend in the meanwhile and can't reproduce the issue anymore.
In the end I'm sure that the heatblock was magnetic and I don't know why exactly.
Maybe you can add the following hint to the README:

If that issue occurs, add the following gcode the _Home_Z-macro before lifting the nozzle:
`G91`
`G0 X10 F{z_drop_feedrate}`
`G90`

jlas1, 2022-03-04:
thanks for the feedback, i will keep that code in mind if this issue is more widespread.
Glad that it's working ok now.

closed 2022-03-04 by jlas1