jlas1/Klicky-Probe #61: Mosquito mount for Voron 0.1

DorjeDorf, 2022-02-21:
Hi,

I know they aren't very popular, but there are still some of us out there who like our Mosquito's and want to keep using them :)

Is there any chance for a miniAB cowling version for the Mosquito with the embedded holder? I see you have one for the Dragon and BMO Dragonfly. 

The voron repo still hosts the Mosquito one. 

Thanks!

jlas1, 2022-02-21:
Sure, point me to the CAD of the Mosquito one and i'll carve it up and put it here.

DorjeDorf, 2022-02-22:
You actually killed me right there. I was 100% sure it was in the normal cad (there's a cowling_skeeter object), but upon double checking - it seems it's not it even though it looks like that's it. I'll double check and will update.

DorjeDorf, 2022-02-22:
Confirmed. It's the skeeter cowling object within the regular v0.1 cad.

jlas1, 2022-02-23:
Done, check the latest CAD and V0 STL files

closed 2022-02-23 by jlas1

DorjeDorf, 2022-02-24:
Rockstar! I'll print it over the weekend and check it out. Can't wait! :)

DorjeDorf, 2022-02-28:
I printed it out - looks awesome. Just one comment. 
The wall where the probe mount latches on is very thin - not like the other cowlings.

Attaching a picture that shows the Skeeter and BMO next to each other - this will show the difference. 

[image]

jlas1, 2022-02-28:
let me look into that, thanks for the feedback

DorjeDorf, 2022-03-01:
Would be awesome, thank you. 
I also need to figure out what is different between our printers. 
I mounted what I have today - the holder is too recessed for the cowling mount to go far enough to pick up the probe. I tried simulating the mount you have on a video - it goes a touch further back, but I don't want to believe that the 3mm more would make so much of a difference.

jlas1, 2022-03-02:
Solved

DorjeDorf, 2022-03-03:
Awesome! Now I just have to wait for Github to show the new files ;)

jlas1, 2022-03-04:
The files should be there already.
It was this commit https://github.com/jlas1/Klicky-Probe/commit/34d59061ca79e80f7a6e7d7539c8eba99431623d