jlas1/Klicky-Probe #62: Quad gantry level Fails

mochoandre, 2022-02-21:
Hi All,
beeen beeen working ok for some mnonths and I Updated Klipper and klipper_z_calibration in last days and since then my Voron 2.4 350 has been a very big paperweight. 

I have updated config files like the new examples in Klicky probe Github.
I can do home and my printer moves perfectly in every axes.
When I do a QUAD_GANTRY_LEVEL print head starts to rattling very loud more or less in the same position and it fails.
If I use klipper_z_calibration config examples I can do all home and QGL procedures but when it starts to print imidieatly starts to loose steps in XY axes.

What can it be?

jlas1, 2022-02-21:
That is weird, but out of scope of klicky.
Have you tried in #voron_2_question on Discord, they might help you better there.

I would guess it's something mechanical, if you sense any binding when moving by hand.
If that is ok i might try to increase the AB motors current.

jlas1, 2022-03-02:
Hi, any update?

mochoandre, 2022-03-03:
Sorry, 
It looks like it was stealthchop. don't know why but after update on klipper (same time as z_calibration update) it start to do that erroneos behaviour. whats strange was that it would only happen when I was going to pick Klicky to do QGL. It homed nice and if i was moving manualy after it would work as expected.

Regards, thank you and sorry to bother you but I was not really getting to problem

jlas1, 2022-03-04:
well, glad you found it. cheers.

closed 2022-03-04 by jlas1