jlas1/Klicky-Probe #64: Bed mesh -Probe triggered prior to movement-

GN1G, 2022-02-23:
Hello,

my QGL works normal and without problems.
But if I run a Bed_Mesh_Calibration, it probes 3-4 points and then says "Probe triggered prior to movement"

Any ideas what ca cause this?

best regrads
GN1G

jlas1, 2022-02-23:
Most likely a cabling issue, check the crimps of your cable.
It can also be a broken wire on the chain, go to the location that you get the "Probe triggered prior to movement" on bed mesh and do a probe_accuracy to see if you manage to trigger it.

GN1G, 2022-02-23:
> Most likely a cabling issue, check the crimps of your cable. It can also be a broken wire on the chain, go to the location that you get the "Probe triggered prior to movement" on bed mesh and do a probe_accuracy to see if you manage to trigger it.

Thank you, I'll check it.

DorjeDorf, 2022-02-24:
What temps are you running the mesh at? I'd assume you have your bed piping hot as well as the nozzle? 
I found that with crappier magnets, you need to drop the temp on the nozzle to 130-160C to have the magnets not be impacted by the temps and lose connectivity.

jlas1, 2022-02-24:
I never noticed that, the magnets above more or less 80 degrees they start to lose strenght, but that is very hard to happen, at least with a silicone sock on the heat block.

GN1G, 2022-02-25:
> What temps are you running the mesh at? I'd assume you have your bed piping hot as well as the nozzle? I found that with crappier magnets, you need to drop the temp on the nozzle to 130-160C to have the magnets not be impacted by the temps and lose connectivity.

The problem came from the stepper cable, one crimp contact was not tight so in some position it triggered the clicky probe contact.

DorjeDorf, 2022-02-25:
> I never noticed that, the magnets above more or less 80 degrees they start to lose strenght, but that is very hard to happen, at least with a silicone sock on the heat block.

Unless you run your mesh on a bed at 100-110C and have a small dive height. Then, the combined heat can impact accuracy. Usually it appears the longer the mesh runs. I know of a couple more people who observed the same effect. We were all meshing for ABS printing so the bed was above 100C. Dropping hotend temp around 130C was enough to do a full mesh :)

But here, seems it was the cable. So good news all around ;D

jlas1, 2022-03-02:
as it's solved, closing the issue.

closed 2022-03-02 by jlas1

HussamAlsahafi, 2023-03-17:
> 

Which stepper motor ?