jlas1/Klicky-Probe #70: V0.1 questions

dmnkhhn, 2022-03-02:
I’ve assembled the Klicky probe today and I have some questions/remarks that might help improve the already excellent instructions even further. 
I had planned to use the automatic Z calibration and it wasn’t too obvious that this isn’t possible with the V0.1 besides clearly stating that it needs the nozzle to trigger the endstop. I was hoping the Klicky probing would do the job. It’s probably not really necessary with such a small bed but it sounded too good. 

What’s the benefit of having the Klicky as the Z endstop?
If I use it that way, what are the next steps before I start my first print? Do I have to adjust my Z offset?

Is the z_offset value of the probe affected by printed parts tolerances and should be individually measured? What are the measuring points? 

What does “probe_accuracy” and “probe_calibrate” do?

The instructions end with the “first test” section. What are the next steps? I have a working Klicky probe but I’m kind of lost what I can actually do with it. 

Thank you for the work done and the work that will be done!

jlas1, 2022-03-02:
Let me try to answer your questions:
What’s the benefit of having the Klicky as the Z endstop?
Allows changing the bed surface or sheet without recalibrating.

If I use it that way, what are the next steps before I start my first print? Do I have to adjust my Z offset?
Yes, you need to adjust the Z offset with probe_calibrate.

Is the z_offset value of the probe affected by printed parts tolerances and should be individually measured? What are the measuring points?
The Z_offset is measured normally on the center of the bed, it is only measured when homing Z, it is not affected by what you are printing, more by at what temperature.

What does “probe_accuracy” and “probe_calibrate” do?
Check the klipper docs, probe_accuracy tests and displays the probe accuracy, probe_calibrate sets the probe z_offset.

The instructions end with the “first test” section. What are the next steps? I have a working Klicky probe but I’m kind of lost what I can actually do with it.
now that you have a working probe, use screws_tilt_calculate to help to adjust the bed screws, you can also do a bed mesh, both options are explained on the klipper docs, much better that by me here. 

https://www.klipper3d.org/

dmnkhhn, 2022-03-03:
Thanks for taking the time to answer my questions, I hope I find the time in the next week to create a pull request with some additions to the instructions.

I will probably link to the parts of the Klipper docs in the relevant places so it's clear where to find more information. I wouldn't have looked at the Klipper docs to find more information about what I thought to be Klicky macros.

Maybe I explained wrong, I understood the `z_offset` (set to 14.5) as the hardware determined distance between the nozzle tip and the switch which might vary (magnets not fully pressed in, printed parts tolerances etc.)

jlas1, 2022-03-03:
The klicky macros, wrap around klipper functions, making sure that the probe is attached and taking care of the attach and dock it has no new functionality.
That being said, please submit a PR if you can, it would be welcomed.
Regarding the probe z_offset, it sometimes change a bit after a few prints, normally it is not a problem, if it is used as Z enstop, then you should calibrate it again after a few prints and printing hours.

closed 2022-03-03 by jlas1