jlas1/Klicky-Probe #77: Rest mount for the DoubleT mod?

Z3ytrox, 2022-03-24:
What mount is used for the doublet mod? Like the mount where the switch rests on when its not used.

jlas1, 2022-03-24:
That is a good point, that info is missing.

You can use these https://github.com/jlas1/Klicky-Probe/blob/main/Printers/ZeroG/Mercury_One/Voron_Toolhead_STL/Dock_mount_Ender5.stl or even these https://github.com/jlas1/Klicky-Probe/tree/main/Printers/Voron/v1.8_v2.4_Legacy_Trident#mounting-options (with some small changes) depends on the distance from the extrusion

Z3ytrox, 2022-03-24:
Will the first link work with the rapido uhf eva mount?

jlas1, 2022-03-24:
maybe, it's adjustable on the z

Z3ytrox, 2022-03-24:
But does it have enough adjustment for this?

jlas1, 2022-03-24:
if it does not, its easy to expand.

Z3ytrox, 2022-03-24:
Ah okay thats awesome! Does it need to be printed in abs?

jlas1, 2022-03-24:
for dev and open printer, no. for an enclosured printer, it would be better

Z3ytrox, 2022-03-24:
Ah okay thats understandable. So petg is prefered over pla then? I want to check it with petg first before I let someone print it in abs for me.

jlas1, 2022-04-08:
PETG should be enough for testing, for long term usage, ABS is more tested and it works.

closed 2022-04-08 by jlas1