jlas1/Klicky-Probe #84: Probe Not Attaching

Alexander-T-Moss, 2022-04-03:
I was finding the magnet in the dock was too strong, and thus the probe was not being pulled out. So I've made a PR with a modified dock that has the magnet slightly further away in the dock from the one on Klicky and found it to work better. I put a video below showing the issue.

https://user-images.githubusercontent.com/54496326/161437432-c8cde412-2bc7-4ae0-91a5-c66492121465.mov

jlas1, 2022-04-05:
i'll close this for now, if there is any issue with the new dock, we can open it again.

closed 2022-04-05 by jlas1

jangrewe, 2022-06-22:
I have the same issue with `Probes/UnklickyNG/STL/KlcikyNG_Probe_Dock.stl`, the current version. My quick fix to get it to work, so that i can print better fixes, was stuffing a bit of cardboard into the magnet hole to keep it further away from the probe's back magnet.