jlas1/Klicky-Probe #87: LGX Lite

ModMike, 2022-04-06:
Can you add an LGX Lite mount to the V0.1 mix? Preferably one that works wit the existing carriage mount so we can use the metal ones and/or avoid having to re-belt the the printer?

Thank you.

jlas1, 2022-04-07:
Can you point me to the LGX Lite cowling that you are referring to?
The only one i know already has support for the integrated Klicky.

ModMike, 2022-04-07:
[Here you go](https://github.com/camerony/VoronCustom/tree/main/MiniASLGXerpaLiteMount). I don't like that back plate and was hoping to integrate with the metal [V0.1 MGNC carriage mount](https://www.aliexpress.com/item/1005003448901997.html?spm=a2g0o.order_list.0.0.21ef18027ZT92O). 

I may take a crack at it myself but I think it would be nice to include and LGX lite extruder option in your GitHub.

jlas1, 2022-04-08:
I believe that that cowling is already in the repository, here https://github.com/jlas1/Klicky-Probe/blob/main/Printers/Voron/v0/v0_STL/Serpa_Mini_Dragon_Cowling_wKlicky.stl.
The LGX mount on top of it needs no modification.

jlas1, 2022-04-23:
gonna close this issue.

closed 2022-04-23 by jlas1

ModMike, 2022-04-23:
I went ahead and redesigned it the way I wanted. Thanks for blazing the trail!
[image]

jlas1, 2022-04-23:
Great that it worked out for you, with what probe did you end up using, i don't see the integrated klicky cutout on that image.

ModMike, 2022-04-23:
I went with my custom version of the zero click. Much smaller and cleaner all around.

[image]

jlas1, 2022-04-23:
looks good, thanks for the feedback.

ModMike, 2022-04-23:
Thank you for all your hard work. I would never had been able to do this without everything you did before.

GadgetAngel, 2022-04-23:
> I went with my custom version of the zero click. Much smaller and cleaner all around.
> 
> [image]

Where do you have this on the web?

ModMike, 2022-04-23:
It's not released yet. I'm still testing. I just finished printing my first set and am making adjustments. I will post when done.

ModMike, 2022-04-29:
In Beta, a few people printing, will release next weekend.

[image]

Rhastlyn, 2022-05-06:
> In Beta, a few people printing, will release next weekend.
> 
> [image]

Need any additional testers?  I'm thinking of this configuration for a rebuild and this looks promising!