jlas1/Klicky-Probe #90: Docking speed in attach_probe is variable_travel_speed

sporkus, 2022-04-14:
In line 199 in klicky-macros.cfg:
```
		G0 X{docklocation_x|int - attachmove2_x|int} Y{docklocation_y|int - attachmove2_y} F{travel_feedrate}
```

The toolhead is currently traveling to the dock location at full travel speed if `variable_attachmove2_x/y` is 0 (default klicky-variable.cfg). `variable_dock_speed` and `variable_release_speed` have no effect. The following change will prevent that from happening.
```
		G0 X{docklocation_x|int - attachmove2_x|int} Y{docklocation_y|int - attachmove2_y} F{dock_feedrate}
```

I am guessing that this is unintended behavior because line 200 is travel to the dock at dock_feedrate. Also, `DOCK_PROBE` doesn't slam the toolhead towards the dock at travel_speed.

jlas1, 2022-04-19:
Thanks, let me check the code and get back to you.

jlas1, 2022-05-13:
Solved, thanks.

closed 2022-05-13 by jlas1