jlas1/Klicky-Probe #92: improvement for failed attaching probe

legend069, 2022-04-16:
hello,

is it possible to add a script or edit the klipper code for attachable probes ?

for example the most common error is for when it doesn't detect the probe is attached or not.
what would be nice is instead of "Probe triggered prior to movement" it does the docking movements and then trys to re attach.
or after it attaches the probe it just checks if it is in the open state and if not it will try again?

thanks.

jlas1, 2022-04-19:
Hi, that might be possible, in my experience however, if the probe does not attach at the first time, it probably won't do on the second one. Most of those errors are probably cabling issues that might be disguised by the second attempt.

But i'll leave this issue open to see if that may be an option.

legend069, 2022-04-27:
I've made this one "check_if_attached"
prob not the best but it's decent as it is

[tempmacrofile.txt](https://github.com/jlas1/Klicky-Probe/files/8570309/tempmacrofile.txt)

jlas1, 2022-11-22:
closing this now