jlas1/Klicky-Probe #94: ! horizontal_move_z can't be less than probe's z_offset

jdlarry, 2022-04-22:
I have done z offset calibration it is 7.7 and i have the horizontal z move set at 10 and when i do g32 it picks up the klicky probe then stops with the above message. need help. thanks

## Voron Design VORON2 250/300/350mm BigTreeTech OctoPus V1 TMC2209 UART config

## *** THINGS TO CHANGE/CHECK: ***
## MCU paths                            [mcu] section
## Thermistor types                     [extruder] and [heater_bed] sections - See 'sensor types' list at end of file
## Z Endstop Switch location            [safe_z_home] section
## Homing end position                  [gcode_macro G32] section
## Z Endstop Switch  offset for Z0      [stepper_z] section
## Probe points                         [quad_gantry_level] section
## Min & Max gantry corner postions     [quad_gantry_level] section
## PID tune                             [extruder] and [heater_bed] sections
## Thermistor types                     [extruder] and [heater_bed] sections
## Probe pin                            [probe] section
## Fine tune E steps                    [extruder] section

[mcu]
##  Obtain definition by "ls -l /dev/serial/by-id/" then unplug to verify
##--------------------------------------------------------------------
serial: /dev/serial/by-id/usb-Klipper_stm32f446xx_14002A000650534E4E313020-if00

restart_method: command
##--------------------------------------------------------------------

[printer]
kinematics: corexy
max_velocity: 300  
max_accel: 4000             #Max 4000
max_z_velocity: 15          #Max 15 for 12V TMC Drivers, can increase for 24V
max_z_accel: 350
square_corner_velocity: 5.0

#####################################################################
#   X/Y Stepper Settings
#####################################################################

##  B Stepper - Left
##  Connected to MOTOR_0
##  Endstop connected to DIAG_0
[stepper_x]
step_pin: PF13
dir_pin: PF12
enable_pin: !PF14
rotation_distance: 40
microsteps: 32
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
endstop_pin: PG6
position_min: 0
##--------------------------------------------------------------------

##  Uncomment below for 250mm build
#position_endstop: 250
#position_max: 250

##  Uncomment for 300mm build
#position_endstop: 300
#position_max: 300

##  Uncomment for 350mm build
position_endstop: 350
position_max: 350

##--------------------------------------------------------------------
homing_speed: 25   #Max 100
homing_retract_dist: 5
homing_positive_dir: true

##  Make sure to update below for your relevant driver (2208 or 2209)
[tmc2209 stepper_x]
[... 430 lines cut ...]
    
    SAVE_GCODE_STATE NAME=STATE_PRINT_END
    
    M400                           ; wait for buffer to clear
    G92 E0                         ; zero the extruder
    G1 E-2.0 F3600                 ; retract filament
    
    TURN_OFF_HEATERS
    
    G90                                      ; absolute positioning
    G0 X{x_safe} Y{y_safe} Z{z_safe} F20000  ; move nozzle to remove stringing
    G0 X{th.axis_maximum.x//2} Y{th.axis_maximum.y - 2} F3600  ; park nozzle at rear
    M107                                     ; turn off fan
    
    BED_MESH_CLEAR
    RESTORE_GCODE_STATE NAME=STATE_PRINT_END
	
	
##---------------------------------------------------------------------------------------------------

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe]
#*# z_offset = 7.700

jlas1, 2022-04-23:
Well, something is strange, can restart your pi and test again. If it does not work, paste here your klipper klippy.log

jdlarry, 2022-04-24:
I found it I didn't have a horizontal move z in my printer.cfg only in klicky probe variables.

jlas1, 2022-05-03:
I believed we have sorted this already, can you confirm?

jlas1, 2022-05-10:
closing this issue.

closed 2022-05-10 by jlas1