jlas1/Klicky-Probe #97: Voron 0 Revo Cowling

jplekkerkerker, 2022-05-02:
Will there be a Voron 0.1 Revo Cowling with integrated klicky-mount?

jlas1, 2022-05-02:
yeah, as soon as it's released, or has it been already?

Grooviee, 2022-05-02:
It's released on the Voron 0-github: https://github.com/VoronDesign/Voron-0/tree/Voron0.1/STLs/Toolheads/Mini_Afterburner/Revo_Voron_Toolhead_DD

jlas1, 2022-05-03:
There is no CAD, without that, i cannot do a klicky version.

Munkee82, 2022-05-04:
https://e3d-online.zendesk.com/hc/en-us/articles/4444939320093-Revo-Voron-Reference-CAD

jlas1, 2022-05-04:
that is the hotend cad, I need the cowling cad for the v0. 1.

Munkee82, 2022-05-04:
This one?
https://github.com/KurioHonoo/Mini-AfterSherpa/blob/main/CAD/MiniAS_Revo_Voron_Base.step

jlas1, 2022-05-07:
This is for the AfterSherpa, but i can mod that one.

Munkee82, 2022-05-10:
That would be cool.  I use the mini AS anyway.

closed 2022-05-17 by jlas1