jlas1/Klicky-Probe #98: First probe value is always off

QuestionDevelopment, 2022-05-02:
Good day.  Thank you very much for this amazing project.  I am encountering a problem where I am unable to complete QUAD_GANTRY_LEVEL (without increasing retry_tolerance from 0.0075 which is the default Voron v2 setting) because the first reading I get from the probe is always significantly off.

Sample output (top is most recent)

### Quad Gantry
```
10:03:04 
// Probe samples exceed tolerance. Retrying...
10:03:04 
// probe at 50.000,275.000 is z=6.970000
10:03:03 
// probe at 50.000,275.000 is z=6.966250
10:03:01 
// probe at 50.000,275.000 is z=6.963750
10:03:00 
// Probe samples exceed tolerance. Retrying...
10:03:00 
// probe at 50.000,275.000 is z=6.960000
10:02:58 
// probe at 50.000,275.000 is z=6.937500
10:02:51 
// probe at 50.000,25.000 is z=10.201250
10:02:49 
// probe at 50.000,25.000 is z=10.201250
10:02:48 
// probe at 50.000,25.000 is z=10.200000
10:02:46 
// probe at 50.000,25.000 is z=10.200000
10:02:45 
// probe at 50.000,25.000 is z=10.198750
10:02:43 
// Probe samples exceed tolerance. Retrying...
10:02:43 
// probe at 50.000,25.000 is z=10.196250
10:02:42 
// probe at 50.000,25.000 is z=10.178750
10:02:36 
// probe: open
10:02:34 
// Attaching Probe
10:02:34 
// probe: TRIGGERED
10:02:33 
// QG Level
10:02:33 
$ quad_gantry_level
```

### Probe Accuracy

```
10:09:56 
// probe accuracy results: maximum 7.805629, minimum 7.781879, range 0.023750, average 7.800629, median 7.803129, standard deviation 0.006869
10:09:56 
// probe at 83.000,297.000 is z=7.805629
10:09:55 
// probe at 83.000,297.000 is z=7.805629
10:09:53 
// probe at 83.000,297.000 is z=7.805629
10:09:52 
// probe at 83.000,297.000 is z=7.804379
10:09:50 
// probe at 83.000,297.000 is z=7.804379
10:09:49 
// probe at 83.000,297.000 is z=7.801879
10:09:47 
// probe at 83.000,297.000 is z=7.800629
10:09:46 
// probe at 83.000,297.000 is z=7.799379
10:09:44 
// probe at 83.000,297.000 is z=7.796879
10:09:43 
// probe at 83.000,297.000 is z=7.781879
10:09:39 
// PROBE_ACCURACY at X:83.000 Y:297.000 Z:25.000 (samples=10 retract=3.000 speed=5.0 lift_speed=5.0)
10:09:39 
// probe: open
10:09:37 
// Attaching Probe
10:09:37 
// probe: TRIGGERED
10:09:36 
$ probe_accuracy
```

This effect is always present/very reproducible.  I've done similar tests at least 50-60 times. I rebuilt my hotend assembly to try and eliminate any slop.  I also rebuilt the klicky probe 3 times (each time with new switch and assembled a little more exact) with no change.  I tested at 6 points around the bed to confirm it was not the bed.  The linear rail is an LDO MGN12...as far as I can tell that is the only place where any slop is present...however is is extremely small and I can only feel it with a rocking motion/not a upwards pressure motion.

I'm 100% stuck at this point.  I would really appreciate your help. Thanks.

QuestionDevelopment, 2022-05-02:
I just sent a friend request on Discord...its a famous guitar players name.  Don't feel obligated to accept...I just wanted to provide that channel of communication if you found it more convenient.

jlas1, 2022-05-02:
i will also answer it here, that is a effect of the z motion system, normally named backlash.

QuestionDevelopment, 2022-05-02:
Just to provide an update.  I think the backlash is coming from my z belts (voron 2).  Here is another person with a similar error report [ https://github.com/Klipper3d/klipper/issues/2024 ].

On top of trying to remove all backlash (try and move/eliminate all movement) adjusting samples_tolerance to a higher value to eliminate the errors