jlas1/Klicky-Probe #100: V0.1 Bed flex under probe pressure

gsezz, 2022-05-05:
When I'm using mesh bed leveling, my Z-offset is alsways about 0.1mm to close in the front, and 0.1mm to far in the back, with a perfect linear gradient back to front. No matter how scewed the bed actually is. I think this is because the froce needed to actuate the probe makes the bed bend more in the front than in the back. 
I already upgraded to the kirigami bed, and used an switch with the lowest actuation force I could find (0.7N). Didin't anybody else have this problem? 
Adding a lever to the switch brought it down to about 0.05mm difference front t o back, wich works, but still isn't pretty. And the lever is horrible for the overall accuracy.

jlas1, 2022-05-06:
I would suspect that you are using klicky on a V0(.1).
There were some users that had this difficulty, most don't see it.
However, there is a good news, i have a beta unklicky probe that triggers on 30g of force i can post it here, let me clean it up a bit.

jlas1, 2022-05-06:
Here it is, enjoy, [Unklicky_forV0.zip](https://github.com/jlas1/Klicky-Probe/files/8638677/Unklicky_forV0.zip) similar instructions on how to assemble here https://github.com/majarspeed/Unklicky.
The wires on the cable ducts should be without insulation

closed 2022-05-06 by jlas1