jlas1/Klicky-Probe #101: Y homing for Y oriented probe ignores _HOME_Y

falo2k, 2022-05-07:
In the homing_override, the y homing section that triggers if attachmove_y ==0 doesn't use the same branching as further down the code.  As a result, for y oriented docks it will never call a _HOME_Y macro if one is defined, only ever calling a G28 Y0.  Confused the hell out of me while I was setting up my sensorless homing :)  Will submit a PR.

falo2k, 2022-05-07:
I'm now wondering if that section also needs to consider attachmove2_y when deciding on the y-first homing approach

jlas1, 2022-05-07:
good question, need to document the moves better.

falo2k, 2022-05-08:
I'd give it a go, but I feel like it would benefit from a gif, and I'm not that talented :)  Once I'm done swearing at my current build I'll have a look.

Is the thinking behind having a y home first for a y-oriented probe that it will avoid it accidentally picking up the probe early on XY homing?

jlas1, 2022-05-08:
it's more to avoid hitting the dock with the probe on initial homing.

falo2k, 2022-05-08:
Ah, gotcha - with the assumption that the dock is mounted along sides rather than back of the printer?  In which case it may need some rethinking then as for the case where I've got side entry but it's attached to the rear (euclid mount, mounted on rear gantry at back left, probe enters right to left) it can still hit the dock if it homes Y first (with a Y-based attachmove).  Maybe better as a home_y_before_x config flag rather than trying to infer it from the movements?

jlas1, 2022-06-23:
solved, now there is a variable (variable_override_homing) to control the homing order.

closed 2022-06-23 by jlas1