jlas1/Klicky-Probe #119: Goal behind klicky probe

reparator, 2022-06-05:
Just a question:
What is the intention or the goal behind it? Which problem does this klicky probe solve? I could not really find that out in the documentation.
On a Voron you normally have a fixed inductive sensor for z-direction. Now it is variable. The tolerances should by higher with that. So what is the advantage?
Need some clarification.
Regards
Stefan

reparator, 2022-06-05:
After looking at that video: [https://www.youtube.com/watch?v=oQYHFecsTto&t=2s](url) now it is clear for me.

closed 2022-06-05 by reparator

jlas1, 2022-06-06:
The goals or objectives are on the initial Readme.md, i have highlighted the most important when coming from an inductive sensor:

    drop in replacement for Omron TL-Q5MC2 or PL-08N2 (you don't need to replace the toolhead), replacement of BLtouch probes
    soldering not required
    minimal adjustments required
    be able to detect all the print surfaces
    be as close to the hotend tip as possible
    highly repeatable and accurate probes
    **less temperature variations**
    **no melting of its parts**
    **cheap to build**
    reuse spare parts if possible