jlas1/Klicky-Probe #122: z-hob disabled, bed still crashes on low position when homing

ulk4, 2022-06-12:
I installed klicky-probe on a Qidi i-mate. Dock is mounted on upper front profile.
i-mate is a printer, where the bed falls down to bottom position (Z max), when motors are off.
I am using klicky-probe as Z-Endstop
When the printer is homing, it still tries to move the bed to a safe_z positon, but this is not possible, since the bed is already at lowest position. Z-hob is disabled.
I am not good in reading macros, so i could not really find out, where it happens and what is the problem.

The relevant configs are
klicky-variables.cfg
```
## only the modified lines 
variable_safe_z:                 7
variable_enable_z_hop:       False
variable_max_bed_y:            200
variable_max_bed_x:            270
variable_z_endstop_x:            0
variable_z_endstop_y:            0	
variable_docklocation_x:       266
variable_docklocation_y:        -7
Variable_dockmove_x:           -40
Variable_attachmove_x:           0
Variable_attachmove_y:         -30
```

printer.cfg
```
## only the changes made for klicky-probe

[include klicky-probe.cfg]

[stepper_z]
endstop_pin: probe:z_virtual_endstop
#position_endstop: 205.030			# removed for klicky-probe

[probe]
pin=^PE3
x_offset: 0
y_offset: -23
z_offset: 4.455
speed: 2.5
samples: 3
samples_result: median
sample_retract_dist: 2.0
samples_tolerance: 0.01
samples_tolerance_retries: 5
```

and the debug log from klipper terminal
```
20:48:21	probe: TRIGGERED
20:48:20	probe: TRIGGERED
20:48:18	Dock_Probe moving away from the dock to G0 X226 Y23 F4500
20:48:17	Dock_Probe moving from the dock to G0 X226 Y-7 F4500
20:48:17	Dock_Probe moving to the dock with G0 X266 Y-7 F3000
20:48:15	Dock_Probe moving near the dock with G0 X266 Y23 F12000
20:48:15	Dock_Probe going to dock probe
20:48:15	Dock_Probe Axis homed
20:48:15	Docking Probe
20:48:15	homing_override probe no longer required, docking probe
20:48:14	_Home_Z toolhead too low, raising it to 7.0mm
20:47:43	_Home_Z Homing Z G28 Z
20:47:43	_Home_Z moving to Z endstop position G0 X134.0 Y124.5 F12000
20:47:43	_Home_Z Axis homed
20:47:43	probe: open
20:47:42	Attach_Probe moving to a safe Z position: G0 Z7 F1200
20:47:41	Attach_Probe moving from the dock to G0 X266 Y23 F4500
20:47:41	Attach_Probe moving to the dock with G0 X266 Y-7 F3000
20:47:41	Attach_Probe moving to the dock with G0 X266 Y-7 F3000
20:47:39	Attach_Probe moving near the dock with G0 X266 Y23 F12000
20:47:38	Attach_Probe toolhead too low, raising it by 7mm, duplicate?
20:47:38	Attach_Probe toolhead too low, raising it by 7mm
20:47:37	Attach_Probe going to attach probe
20:47:37	moving to a safe Z distance
20:47:37	Attaching Probe
20:47:37	homing_override probe configured as a virtual Z endstop attaching probe
20:47:37	homing_override backing off Y endstop, G0 Y193.0 F12000
20:47:32	homing_override Homing Y G28 Y0
20:47:32	homing_override backing off X endstop, G0 X258.0 F12000
20:47:26	homing_override Homing X G28 X0
20:47:26	homing_override toolhead too low, raising it by 0mm
20:47:26	homing_override Z not homed, setting position as X=Y=Z=0
20:47:26	homing_override z_hop disabled
20:47:25	homing_override goint to home all axes
20:47:25	probe: TRIGGERED
20:47:25	Homing Z
20:47:25	Homing Y
20:47:25	Homing X
20:47:25	Z not homed, forcing full G28
20:47:25	G28
```
I think the bed move in positive Z direction is happening at 20:47:38, but it is scrolling quite fast.

Btw, when I apply the klicky-probe manually before homing, it is working without the collision move.

For the moment as a workaround, I lowered safe_z to only 7mm and placed some foam on the bottom of the printer, so the bed does not fall down all the way and the collision is a soft one.
Any idea, why `variable_enable_z_hop:       False` is not working as expected?

And last not least, Docking, attaching and bed mesh calibration are working fine :-)

jlas1, 2022-06-23:
I see the issue, it needs to be able to move to attach if the Z is too low.
Try this macro:
https://raw.githubusercontent.com/jlas1/Klicky-Probe/z_hop_tests/Klipper_macros/klicky-variables.cfg
It should work as expected, it is on a new branch.

ulk4, 2022-06-26:
I used the macro [https://raw.githubusercontent.com/jlas1/Klicky-Probe/z_hop_tests/Klipper_macros/klicky-macros.cfg](url) from
z-hob_tests branch. 
Now, when the printer is homing, before homing x any y, the bed is moved up by `variable_safe_z`. Later on, before attaching the probe, the bed is lowered again the same amount, which is now possible. I guess this is the expected behaviour.
At least, it does not try to move the bed downwards anymore in the beginning, when it is already at bottom position.
Looks good to me, I can take out the foam again :-).
Thanks.

jlas1, 2022-06-27:
Can you enable the debug variable on klicky-variables.cfg and do a homing please?
Would like to find out why is it moving like that.

ulk4, 2022-06-27:
Sure. Here is the debug log

**07:42:53	probe: TRIGGERED
07:42:52	probe: TRIGGERED
07:42:50	Dock_Probe moving away from the dock to G0 X226 Y23 F4500
07:42:49	Dock_Probe moving from the dock to G0 X226 Y-7 F4500
07:42:49	Dock_Probe moving to the dock with G0 X266 Y-7 F3000
07:42:47	Dock_Probe moving near the dock with G0 X266 Y23 F12000
07:42:47	Dock_Probe going to dock probe
07:42:47	Dock_Probe Axis homed
07:42:47	Docking Probe
07:42:47	homing_override probe no longer required, docking probe
07:42:46	_Home_Z toolhead too low, raising it to 7.0mm
07:42:21	_Home_Z Homing Z G28 Z
07:42:21	_Home_Z moving to Z endstop position G0 X134.0 Y124.5 F12000
07:42:21	_Home_Z XY Axis homed
07:42:21	probe: open
07:42:21	Attach_Probe moving to a safe Z position: G0 Z7 F1200 from -0.029827
07:42:19	Attach_Probe moving from the dock to G0 X266 Y23 F4500
07:42:19	Attach_Probe moving to the dock with G0 X266 Y-7 F3000
07:42:19	Attach_Probe moving to the dock with G0 X266 Y-7 F3000
07:42:18	Attach_Probe moving near the dock with G0 X266 Y23 F12000
07:42:16	Attach_Probe toolhead too low, raising it by 7mm, duplicate?
07:42:16	Attach_Probe toolhead too low, raising it by 7mm
07:42:16	Attach_Probe going to attach probe
07:42:16	moving to a safe Z distance
07:42:16	Attaching Probe
07:42:16	homing_override probe configured as a virtual Z endstop attaching probe
07:42:10	homing_override Homing Y G28 Y0
07:42:03	homing_override Homing X G28 X0
07:42:03	homing_override toolhead too low, raising it by 0mm
07:42:03	homing_override Z not homed, setting position as X=Y=0 Z=7
07:42:03	homing_override z_hop disabled
07:42:03	homing_override goint to home all axes
07:42:03	probe: TRIGGERED
07:42:03	Homing Z
07:42:03	Homing Y
07:42:03	Homing X
07:42:03	Z not homed, forcing full G28
07:42:03	G28**

I think it happens at **07:42:03	homing_override toolhead too low, raising it by 0mm**, even when it says it's moving by 0mm. Bed is raised by safe_z.
My guess is, kinematic position is set to Z=7 and then the bed is moved absolutely to safe_z, which is 0 at that time. This results in the first bed raise. BUT: it is only my guess, i'm not good in reading and understanding macros.

jlas1, 2022-06-27:
Can you try now with the main macros?
https://raw.githubusercontent.com/jlas1/Klicky-Probe/main/Klipper_macros/klicky-macros.cfg

You are better at macros than you think, your guess is right.

ulk4, 2022-06-27:
Perfect !
No unnecessary moves any more.
Here is the log with unhomed printer.

```
13:04:05	probe: TRIGGERED
13:04:04	probe: TRIGGERED
13:04:02	Dock_Probe moving away from the dock to G0 X226 Y23 F4500
13:04:00	Dock_Probe moving from the dock to G0 X226 Y-7 F4500
13:04:00	Dock_Probe moving to the dock with G0 X266 Y-7 F3000
13:03:59	Dock_Probe moving near the dock with G0 X266 Y23 F12000
13:03:59	Dock_Probe going to dock probe
13:03:59	Dock_Probe Axis homed
13:03:59	Docking Probe
13:03:59	homing_override probe no longer required, docking probe
13:03:58	_Home_Z toolhead too low, raising it to 7.0mm
13:03:33	_Home_Z Homing Z G28 Z
13:03:33	_Home_Z moving to Z endstop position G0 X134.0 Y124.5 F12000
13:03:33	_Home_Z XY Axis homed
13:03:33	probe: open
13:03:31	Attach_Probe moving from the dock to G0 X266 Y23 F4500
13:03:31	Attach_Probe moving to the dock with G0 X266 Y-7 F3000
13:03:31	Attach_Probe moving to the dock with G0 X266 Y-7 F3000
13:03:29	Attach_Probe moving near the dock with G0 X266 Y23 F12000
13:03:29	Attach_Probe going to attach probe
13:03:29	Attaching Probe
13:03:29	homing_override probe configured as a virtual Z endstop attaching probe
13:03:21	homing_override Homing Y G28 Y0
13:03:18	homing_override Homing X G28 X0
13:03:18	homing_override Z not homed, setting position as X=Y=0 Z=7
13:03:18	homing_override z_hop disabled
13:03:18	homing_override goint to home all axes
13:03:18	probe: TRIGGERED
13:03:18	Homing Z
13:03:18	Homing Y
13:03:18	Homing X
13:03:18	Z not homed, forcing full G28
13:03:18	G28
```

Thanks for your help.
Btw, do you want to add the custom klicky-probe files for the Qidi i-mate to your github? Don't know, if it makes sense, since you need several modifications to the printer in advance. (klipper and a new printhead).

closed 2022-06-27 by jlas1

MrReflex89, 2023-12-27:
Unfortunately, after more than a year the link no longer works. Can someone please tell me where I can find this cfg because I have the same problem with the same printer and have little knowledge of macros to change them myself.
( don´t work: https://raw.githubusercontent.com/jlas1/Klicky-Probe/main/Klipper_macros/klicky-macros.cfg :( )

MrReflex89, 2023-12-27:
Hello everyone, I also have the same problem that if the bed is on the old limit switch and continues to move freely at the beginning, this leads to a collision. I have another problem when a print is finished that the printer switches to save z 10, which would cause a collision with the printed part. Of course I could set the safety height to 200, but when touching it at the start of the print program it would unnecessarily move up and down twice. does anyone have an idea what I can do? I have the latest macros and configs.

klicky-variables.cfg:
[gcode_macro _User_Variables]
variable_verbose:             True    # Enable verbose output
variable_debug:              False    # Enable Debug output
variable_travel_speed:         200    # how fast all other travel moves will be performed when running these macros
variable_move_accel:          1000    # how fast should the toolhead accelerate when moving
variable_dock_speed:            50    # how fast should the toolhead move when docking the probe for the final movement
variable_release_speed:         75    # how fast should the toolhead move to release the hold of the magnets after docking
variable_z_drop_speed:          20    # how fast the z will lower when moving to the z location to clear the probe

variable_safe_z:         	    10    # Minimum Z for attach/dock and homing functions     #orig: 25
# if true it will move the bed away from the nozzle when Z is not homed
variable_enable_z_hop:        False    # set this to false for beds that fall significantly under gravity (almost to Z max)   #orig: True

variable_max_bed_y:            200    # maximum Bed size avoids doing a probe_accuracy outside the bed
variable_max_bed_x:            270    # maximum Bed size avoids doing a probe_accuracy outside the bed

# if a separate Z endstop switch is in
# use, specify the coordinates of the switch here (Voron).
# Set to 0 to have the probe move to center of bed
variable_z_endstop_x:         0        #orig: 1000
variable_z_endstop_y:         0        #orig: 1000

#Check the printer specific documentation on klipper Dock/Undock configuration, these are dummy values
#dock location
variable_docklocation_x:       268    # X Dock position
variable_docklocation_y:         6    # Y Dock position
variable_docklocation_z:      -128    # Z dock position (-128 for a gantry/frame mount)

#The following variables are used if the dock is deployed and retracted via a servo motor
variable_enable_dock_servo:  False    # Set to true if your klicky dock is servo-controlled
variable_servo_name:        'NAME'    # The name of the dock servo defined in printer.cfg under [servo]
variable_servo_deploy:          10    # This EXAMPLE is the value used to deploy the servo fully
variable_servo_retract:         11    # This EXAMPLE is the value used to retract the servo fully (initial_angle in [servo] config)
variable_servo_delay:          250    # This is a delay to wait the servo to reach the requested position, be carefull with high values

#Dock move, final toolhead movement to release the probe on the dock
#it's a relative move
Variable_dockmove_x:           -40
Variable_dockmove_y:             0
Variable_dockmove_z:             0

#Attach move. final toolhead movement to attach the probe on the mount
#it's a relative move
Variable_attachmove_x:           0
Variable_attachmove_y:         -30
Variable_attachmove_z:           0

#Umbilical to help untangle the umbilical in difficult situations
variable_umbilical:          False    # should we untangle the umbilical
variable_umbilical_x:           15    # X umbilical position
variable_umbilical_y:           15    # Y umbilical position

# location to park the toolhead
variable_park_toolhead:      False    # Enable toolhead parking
variable_parkposition_x:       125
variable_parkposition_y:       125
variable_parkposition_z:        30

variable_version:                1    # Helps users to update the necessary variables, do not update if the variables above are not updated

#Below this remark, you normally do not need to configure
#Attach move2
Variable_attachmove2_x:          0    # intermediate toolhead movement to attach
Variable_attachmove2_y:          0    # the probe on the dock
Variable_attachmove2_z:          0    # (can be negative)

#variable_home_backoff_x:        10    # how many mm to move away from the X endstop after homing X
[... 1243 lines cut ...]
    {% if printer['gcode_macro status_homing'] is defined %}
        _KlickyDebug msg="_klicky_status_homing activating the LED STATUS_HOMING"
        STATUS_HOMING
    {% endif %}

[gcode_macro _klicky_status_cleaning]
gcode:
    {% if printer['gcode_macro status_cleaning'] is defined %}
        _KlickyDebug msg="_klicky_status_cleaning activating the LED STATUS_CLEANING"
        STATUS_CLEANING
    {% endif %}

[gcode_macro _klicky_status_meshing]
gcode:
    {% if printer['gcode_macro status_meshing'] is defined %}
        _KlickyDebug msg="_klicky_status_meshing activating the LED STATUS_MESHING"
        STATUS_MESHING
    {% endif %}

[gcode_macro _klicky_status_calibrating_z]
gcode:
    {% if printer['gcode_macro status_calibrating_z'] is defined %}
        _KlickyDebug msg="_klicky_status_calibrating_z activating the LED STATUS_CALIBRATING_Z"
        STATUS_CALIBRATING_Z
    {% endif %}