jlas1/Klicky-Probe #123: Bed Crashing to bottom on V0.1

jeronbernal, 2022-06-16:
I would love any help i can get on this, i feel like i'm completely missing something >_>;;

When testing after homing X/Y the bed will try to lower 25, then lower again. then give me a Move out of Range : 

08:56:44
Move out of range: -879.000 14.000 25.000 [0.000]
08:56:44
Move out of range: -879.000 14.000 25.000 [0.000]
08:56:40
Resetting Z position to zero, duplicate?
08:56:40
moving to a safe Z distance
08:56:40
Resetting Z position to zero
08:56:40
Attaching Probe

My klicky_variable.cfg : 
# This macro was provided by discord user Garrettwp to whom i give my thanks for sharing it with me.
# I have tweaked it a lot.
# They are based on the great Annex magprobe dockable probe macros "#Originally developed by Mental,
# modified for better use on K-series printers by RyanG and Trails", kudos to them.
# That macro as since evolved into a klipper plugin that currently is pending inclusion in klipper, 
# more information here, https://github.com/Annex-Engineering/Quickdraw_Probe/tree/main/Klipper_Macros
# User richardjm revised the macro variables and added some functions, thanks a lot
# by standing on the shoulders of giants, lets see if we can see further
#
# the current home for this version is https://github.com/jlas1/Klicky-Probe 
# the 1000 values below is to give an error instead of doing something wrong, hopefully, this won't be used is a printer larger than 1 meter

[gcode_macro _User_Variables]
variable_verbose:               True  # Enable verbose output
variable_travel_speed:          100   # how fast all other travel moves will be performed when running these macros
variable_dock_speed:            50    # how fast should the toolhead move when docking the probe for the final movement
variable_release_speed:         55    # how fast should the toolhead move to release the hold of the magnets after docking
variable_z_drop_speed:          20    # how fast the z will lower when moving to the z location to clear the probe
    
variable_safe_z:         	    25    # Minimum Z for attach/dock and homing functions
# if true it will move the bed away from the nozzle when Z is not homed
variable_enable_z_hop:        True    # True v0
    
#Dock move
Variable_dockmove_x:             0    # Final toolhead movement to release
Variable_dockmove_y:            40    # the probe on the dock
Variable_dockmove_z:             0    # (can be negative)

#Attach move
Variable_attachmove_x:          30    # Final toolhead movement to Dock
Variable_attachmove_y:           0    # the probe on the dock
Variable_attachmove_z:           0    # (can be negative)
    
variable_max_bed_x:             120   # maximum Bed size avoids doing a probe_accuracy outside the bed
variable_max_bed_y:             120   # maximum Bed size avoids doing a probe_accuracy outside the bed

# if a separate Z endstop switch is in
# use, specify the coordinates of the switch here (Voron).
# Set to 0 to have the probe move to center of bed
variable_z_endstop_x:         0
variable_z_endstop_y:         0

#Check the printer specific documentation on klipper Dock/Undock configuration, these are dummy values
#dock location 
variable_docklocation_x:      121    # X Dock position
variable_docklocation_y:      14    # Y Dock position
variable_docklocation_z:      -128    # Z dock position (-128 for a gantry/frame mount)

#The following variables are used if the dock is deployed and retracted via a servo motor
variable_enable_dock_servo:  False    # Set to true if your klicky dock is servo-controlled
variable_servo_name:        'NAME'    # The name of the dock servo defined in printer.cfg under [servo]
variable_servo_deploy:          10    # This EXAMPLE is the value used to deploy the servo fully
variable_servo_retract:         11    # This EXAMPLE is the value used to retract the servo fully (initial_angle in [servo] config)
variable_servo_delay:          250    # This is a delay to wait the servo to reach the requested position, be carefull with high values

#Dock move, final toolhead movement to release the probe on the dock
#it's a relative move
Variable_dockmove_x:          1000
Variable_dockmove_y:             0
Variable_dockmove_z:             0

#Attach move. final toolhead movement to attach the probe on the mount
#it's a relative move
Variable_attachmove_x:        1000
Variable_attachmove_y:           0
Variable_attachmove_z:           0

#Umbilical to help untangle the umbilical in difficult situations
variable_umbilical:          False    # should we untangle the umbilical
variable_umbilical_x:           15    # X umbilical position
variable_umbilical_y:           15    # Y umbilical position

# location to park the toolhead
variable_park_toolhead:      False    # Enable toolhead parking
variable_parkposition_x:       125
variable_parkposition_y:       125
variable_parkposition_z:        30

variable_version:                1    # Helps users to update the necessary variables, do not update if the variables above are not updated

#Below this remark, you normally do not need to configure
#Attach move2
Variable_attachmove2_x:          0    # intermediate toolhead movement to attach
Variable_attachmove2_y:          0    # the probe on the dock
Variable_attachmove2_z:          0    # (can be negative)

variable_home_backoff:          10    # how many mm to move away from the endstop after homing

variable_override_homing:        ''   # configures what axis to home first
                                      #  '' = default klicky behavior (tries to avoid the hitting the dock
                                      # 'X' = forces X to home first 
                                      # 'Y' = forces Y to home first  

# Do not modify below
gcode:
    {% set Mx = printer['configfile'].config["stepper_x"]["position_max"]|float %}
    {% set My = printer['configfile'].config["stepper_y"]["position_max"]|float %}
    {% set Ox = printer['configfile'].config["probe"]["x_offset"]|float %}
    {% set Oy = printer['configfile'].config["probe"]["y_offset"]|float %}
    {% set Oz = printer['configfile'].config["probe"]["z_offset"]|float %}

    # If x, y coordinates are set for z endstop, assign them
    {% if z_endstop_x != 0 or z_endstop_y != 0 %}
        SET_GCODE_VARIABLE MACRO=_Probe_Variables VARIABLE=z_endstop_x VALUE={ z_endstop_x }
        SET_GCODE_VARIABLE MACRO=_Probe_Variables VARIABLE=z_endstop_y VALUE={ z_endstop_y }

    # if no x, y coordinates for z endstop, assume probe is endstop and move toolhead to center of bed
    {% else %}
        SET_GCODE_VARIABLE MACRO=_Probe_Variables VARIABLE=z_endstop_x VALUE={ (Mx * 0.5) - Ox }
        SET_GCODE_VARIABLE MACRO=_Probe_Variables VARIABLE=z_endstop_y VALUE={ (My * 0.5) - Oy }
    {% endif %}

jlas1, 2022-06-16:
Hi, you need to modify the dockmoves:
Variable_dockmove_x: 1000
Variable_dockmove_y: 0
Variable_dockmove_z: 0

#Attach move. final toolhead movement to attach the probe on the mount
#it's a relative move
Variable_attachmove_x: 1000
Variable_attachmove_y: 0
Variable_attachmove_z: 0

For a V0.1the correct is here:
https://github.com/jlas1/Klicky-Probe/tree/main/Printers/Voron/v0#step-5-klipper-configuration

#Dock move
Variable_dockmove_x:             0    # Final toolhead movement to release
Variable_dockmove_y:            40    # the probe on the dock
Variable_dockmove_z:             0    # (can be negative)

#Attach move
Variable_attachmove_x:          30    # Final toolhead movement to Dock
Variable_attachmove_y:           0    # the probe on the dock
Variable_attachmove_z:           0    # (can be negative)

jeronbernal, 2022-06-17:
Thank you, i copied it over but i didn't get rid of the same attachmove and
dockmove below. it works perfectly now! thank you so much for the klicky,
it's amazing to watch in use.

I had another question regarding docking and attaching between homing and
bed mesh prior to a print.  How can i prevent the klicky from docking
between G28 and bedmesh?   Thank you!!

jlas1, 2022-06-17:
Try this:
G28 X Y
ATTACH_PROBE_LOCK
G28 Z
BED_MESH_CALIBRATE
DOCK_PROBE_UNLOCK

jlas1, 2022-07-04:
closing this issue, can always be opened again

closed 2022-07-04 by jlas1