jlas1/Klicky-Probe #124: Issue with attach probe causing "move out of range"

igetio, 2022-06-17:
this is only after a print is run on my system. When I try to start another without restarting the firmware I get an error like the following when the probe tries to attach. my bed is 350x350x350

`10:05:01
Move out of range: 194.000 420.000 25.000 [36736.676]
10:05:01
Attach_Probe moving near the dock with G0 X14 Y320 F12000`

jlas1, 2022-06-19:
I have not seen that in a long time. can you send me a longer log?
Klicky macros should be immune to G90/G91 issues, but you seem to have one there.
I would assume that if you put a G90 on your print_start, it will start to work, but should not be necessary.

igetio, 2022-06-24:
[klippy (2).log](https://github.com/jlas1/Klicky-Probe/files/8982409/klippy.2.log)
My log is attached and my config is at [https://github.com/igetio/voron24config](https://github.com/igetio/voron24config )

igetio, 2022-07-05:
I was able to resolve the issue seems there was something odd in my initial purge block and placing my start_print logic in a save/restore gcode state has cleared the issue.

closed 2022-07-05 by igetio