jlas1/Klicky-Probe #133: KlickSwipe dosen't fit new MG90s

thbl, 2022-07-14:
it seams that the new MG90s is shorter, could you make your mount fit them ? :)
Sadly i don't have the skills :i but here are cads for them
https://grabcad.com/library/towerpro-mg90s-micro-servo-1
https://grabcad.com/library/mg90d-2
@jlas1 

Also would it not be better to use some kind of rods instead of PTFE ?

jlas1, 2022-07-16:
Can you measure yours, i purchased mines pretty recently.

thbl, 2022-07-16:
@jlas1 also orginals ? as you can see on their website they are lower (do you need any other measurements then on this link?)
https://www.towerpro.com.tw/product/mg90s-3/

jlas1, 2022-07-18:
It should be ok, have you tried to print and fit it
where was your difficulty?

jlas1, 2022-07-18:
Also, you can use a metal pin, most people has PTFE tubes and they also work fine, so YMMV.

closed 2022-07-26 by jlas1