jlas1/Klicky-Probe #137: Directions missing

GaryBoone, 2022-07-26:
The directions are confusing because they appear to stop mid setup. 

The directions in the [readme](https://github.com/jlas1/Klicky-Probe/#readme) are quite specific at first, explaining which files to comment out and how to modify `printer.cfg` to `[include klicky-probe.cfg]`. But then it essentially stops, next saying that "You now need to configure the probe pin, that is printer specific, and the details are on your printer configuration guide." Further, it seems to say there's nothing more to do because no changes are needed to the macros: "Regarding you print start and end macros, with the helper scripts implemented, they do not need to be changed."

However, as far as I've been able to figure out, there are a number of next steps needed, such as changing to `klicky-variables.cfg` to set the location of the Z endstop with `variable_z_endstop_x`.

So: the documentation should continue the basic setup, explaining where to set variables and how to test the setup.

It would also be extremely helpful to suggest basic commands to run as the variables are configured. For example, I think there's a command to go pick up Klicky from the dock. That seems like a basic check on some variables that locate the dock.

Ideally, the directions would have a consistent format of "variable to configure: X" followed by "test with command: Y".

jlas1, 2022-07-26:
the printers setup is where the klicky-variables are detailed along with the rest of the setup process.
Did you check the instructions for your your specific printer, or it has not been created yet?

GaryBoone, 2022-07-27:
What's "the printers setup"? Do you mean the printer.cfg, eg from here: https://github.com/Klipper3d/klipper/tree/master/config ? Or do you mean one of klicky's files in this repo? 

I just found https://github.com/jlas1/Klicky-Probe/tree/main/Printers/Voron/v1.8_v2.4_Legacy_Trident#step-5-klipper-configuration Yes, these are again detailed. That must be what you mean!

Ok, so back to this issue report. The text does say:

    Edit klicky-variables.cfg (there are printers specific recommendations on the printer configuration page) for klicky to operate properly.

But it also has https://github.com/jlas1/Klicky-Probe/tree/main/Printers/Voron/v1.8_v2.4_Legacy_Trident#step-5-klipper-configuration . 

So I think the text at https://github.com/jlas1/Klicky-Probe/tree/main/Klipper_macros#printercfg-configuration would be clearer if it:

1. used numbered steps instead of plain text
2. specified where to look in the repo for your specific printer, 
3. clarified which order setups should be done, Ie, the ordering between https://github.com/jlas1/Klicky-Probe/tree/main/Klipper_macros#printercfg-configuration and https://github.com/jlas1/Klicky-Probe/tree/main/Printers/Voron/v1.8_v2.4_Legacy_Trident#step-5-klipper-configuration For example one of them mentions removing `safe_z_home`; both of them mention editing `klicky-variables.cfg`. Which should be edited when?

Hopefully this is only a small edit to clarify the docs in the repo's frontpage README.md. I realize there's tremendous complexity in keeping all of the moving targets documented. Congrats on the success of Klicky!

jlas1, 2022-11-25:
Hi, did 1 and 2 above, the 3 seems clear to me, on the printer specific configuration, it says "you need to check that before continuing on the build".
So you need to do the generic klipper setup first and then do the specific one, otherwise it's too much variables.

closed 2022-11-29 by jlas1