jlas1/Klicky-Probe #139: MIssing body STL for 3mm magnets

shlagwuk, 2022-08-01:
Hi,
With 3mm height magnets, the body STL is too small and the back part often breaks.

In the CAD file, there is a 3mm body that works better! Is it possible to add it to the rep files ?

In the screenshot below, the 3mm file from cad is enlightened, and the bigger space is visible. 
[image]

poon-holder, 2022-08-05:
The Mini AB cowing mount is also a bugger, keeps knackering the part when inserting the magnets. Are the designs meant for ABS?

Brassn, 2022-08-08:
Sadly I cannot find the better working 3mm body part you mentioned in the CAD step file. Is it possible that you had an older version of the CAD files laying around? Current version is v9.

Also confusing is that BOM clearly prefers 3mm magnets but the parts make it seem like 2.75mm was standard and 3mm was a (less supported?) alternative.

jlas1, 2022-11-22:
Try with the latest parts it should work better.

closed 2022-11-22 by jlas1