jlas1/Klicky-Probe #144: [FR] Publish Klipper_macros.zip as releases for Moonraker update manager

AdamYellen, 2022-08-16:
Would it be possible to publish the Klipper_macros.zip file under Releases to enable Moonraker update manager functionality? I believe the necessary includes from klick-probe.cfg can be done in printer.cfg and all the user variables can be overridden there too, leaving the contents of Klipper_macros.zip untouched.

AdamYellen, 2022-08-16:
Update: looks like it's not quite that simple. Moonraker wants what looks like a custom RELEASE_INFO json file in order to support .zip update management.

Updating from a git repo doesn't require any auxiliary files. Perhaps you could create a repo that only contains the macros?

AdamYellen, 2022-08-16:
For reference, I created a GitHub action that will periodically sync your macros to a new repo which is compatible with Moonraker update manager.

Here is the action file: [sync-push](https://github.com/AdamYellen/Klicky-Probe-macros/blob/main/.github/workflows/sync-push.yml)

Definition in moonraker.conf:
```
[update_manager klicky_probe]
type: git_repo
path: ~/klipper_config/klicky-probe
origin: https://github.com/AdamYellen/Klicky-Probe-macros.git
primary_branch: main
managed_services: klipper
```

Includes in printer.cfg:
```
[include klicky-probe/klicky-variables.cfg]
[include klicky-probe/klicky-macros.cfg]
[include klicky-probe/klicky-bed-mesh-calibrate.cfg]
#[include klicky-probe/klicky-screws-tilt-calculate.cfg]
#[include klicky-probe/klicky-quad-gantry-level.cfg]
[include klicky-probe/klicky-z-tilt-adjust.cfg]
```

Then my overrides in printer.cfg:
```
[gcode_macro _User_Variables]
variable_max_bed_y:           250
variable_max_bed_x:           250
...
Variable_attachmove_x:        0
Variable_attachmove_y:        30
```

AdamYellen, 2022-08-17:
Here's the final project: https://github.com/AdamYellen/Klicky-Probe-macros

AdamYellen, 2022-08-17:
Reopening to ask a question.

I noticed that on the most recent [commit](https://github.com/jlas1/Klicky-Probe/commit/3f77798391da34d09d8dc5f50f65a762805c8897) you updated the zip but the individual config files didn't change. Is the zip kept in sync with the config files?

jlas1, 2022-11-22:
Hi, yes, it is, probably updated the zip just to make sure it was up to date.
Great work, need to take a look at your code.

closed 2022-11-22 by jlas1