jlas1/Klicky-Probe #154: Safe Z doesn't appear to be working on an x/y home

simpat1zq, 2022-09-23:
Before I setup the Klicky macros, I was using safe_z_home, which is now commented out per the instructions. 

However, when I do an x/y home, it is not raising the z, which can potentially cause issues if the nozzle is close to the bed. How do I get that feature back where it raises the z when homing x/y?

jlas1, 2022-11-18:
Can you check the latest macro version and advise if the bug is still there?
also, please include your klicky-variables.cfg

jlas1, 2022-11-25:
should be solved now, if necessary please reopen.

closed 2022-11-25 by jlas1

Eiranya, 2023-01-11:
> should be solved now, if necessary please reopen.

I used Latest macro,if the xyz axes are already homed,run the full G28 command x/y axes will directly goint to home without raising the z,but if it close to the heatbed  when toolhead run to z endstop position the toolhead will hit the z endstop. Moves the Z to safe_z distance only on the first G28 after motors losing power.