jlas1/Klicky-Probe #158: [KlickyNG] AB/SB Mount Cracking When Tightening M2x10 Screws

ndanyluk, 2022-10-05:
Hi, love Klicky and happy to get it installed on my SW, but wanted to flag an issue I saw during assembly.

It seems that the AB/SB mount suffers strain when tightening the M2 screws that hold the magnets in place. On four separate attempts, I managed to crack the AB/SB mount when tightening the right hand screw.  Each time I had not tightened the screw the entire way yet and had to be extremely delicate with tightening on my final successful attempt.

[image]

That thin line across the side is from the plastic splitting.  I used standard Voron part settings (4x perimeters, 5x solid top/bottom layers, 40% infill) but it may help to print with lower infill in that section so the screw doesn't force the plastic apart.

ndanyluk, 2022-10-06:
OK quick update - after 10 repeated failures with varying settings (different flow, xy compensation, drilling out the hole with a 1/16th bit) I decided to jump into the slicer and see what this looks like.

[image]

It seems that with the spec'd settings, this area becomes a solid block of plastic with some gap fill.  It might do better with arachne, but in my case the self-tapping screws exert too much force on these outer walls and split the plastic.  Probably a mix of poor layer adhesion and just the wrong tolerance on the screws, but figured I'd share my experience.  Would probably recommend opening this hole up by .1mm but I'm out of filament to test with unfortunately.

Ramalama2, 2022-10-26:
You have to use definitively arachne. I don't know how the world worked before Arachne anymore 😂

Backstory first:
Tbh, tolerances aren't that easy as you think, designers like me are trying to find a middle ground that works for most people.
But primarily the designer does what works best for himself, because he can selfprint the parts and test directly.
The part with "what works best for most" comes then a lot later, once people start to complain, the first ones then get ignored and if there are more then the tolerances get adjusted.

Now the problem with tolerances relies heavily on flowrate, calibrated extruder, filament width, nozzle...
And everyone or every printer is a little bit different in that aspect.
Including that those self tapping screws are all a little bit different either.
On your particular problem the fan speed, layerhight play a big role either, because it's like an overhang, that hole.

Now back to arachne, the slicers before Arachne with the old engine, can differ a hole up to 0.1mm im worst cases even up to 0.2mm, i seen that and it's extremely horrible on very small holes especially.
Depending on how far they are away from outer wall etc.
I myself try to design holes and the distance of those holes always in multiple of 0.4 or 0.2mm.
And i do that only for the compatibility for the old engine.
For Arachne, i can basically give a fuck to everything and design how i like, because Arachne does thicker/thinner lines and that helps extremely to keep the parts how they are meant.

And as last point, your print looks a bit crappy to me, like not well calibrated etc.... Especially pressure advance.

So as conclusion after the whole writeup, try first to switch to Arachne and tune your printer a bit.
Then if we are lucky it's probably solved for this AND for future prints!

Cheers

jlas1, 2022-11-21:
You may be over-extruding a bit, this happened on the initial revisions, but it seems that it works for most people as is right now, hard to change that with only 1 report.
You can also increase the hole size of that part on a CAD software, should be easy, but if you need, DM me in discord or reopen this issue.

Best Regards and thanks for the issue.

closed 2022-11-21 by jlas1

grimlokason, 2023-01-22:
Hello,

I'm having the same issue, even by under extruding.
I'm also having hard time with a 2 mm screw tap 
I noticed that your hole are 1.5mm while it should be 1.6 according what i saw for M2 screw ( http://www.carbidedepot.com/formulas-tap-metric.htm )

i'll try with an hole of that side and come back to you