jlas1/Klicky-Probe #165: Error in Klicky macro.cfg

BOAndy1985, 2022-11-19:
Homen in z is no longer possible with the new version of macro.  The error says to home x and y first before going to the dock

khumarahn, 2022-11-19:
Same here, I think
```
14:23:07  $ G28
14:23:07  // No axis homed
14:23:07  // moving to a safe Z distance
14:23:07  // X or Y not homed, forcing full G28
14:23:07  // Homing X
14:23:07  // Homing Y
14:23:07  // Homing Z
14:23:07  // probe: TRIGGERED
14:23:12  !! Error evaluating 'gcode_macro Dock_Probe:gcode': CommandError: Must Home X, Y and Z Axis First!
14:23:12  !! Error evaluating 'gcode_macro Dock_Probe:gcode': CommandError: Must Home X, Y and Z Axis First!
​```

khumarahn, 2022-11-19:
I guess that 
```
variable_dock_on_zhome:       True    # docks the probe on Z Homing if not necessary (avoids hitting the bed on some printers                                                                                                                                                                                                  
                                                                                                                                                                                                                                                                                                                               
variable_bypass_probe_docking:        False
```
the procedure is to dock the probe if unnecessary for Z homing, but docking requires Z already homed

khumarahn, 2022-11-19:
Also I wonder what `variable_bypass_probe_docking` is?..

BOAndy1985, 2022-11-19:
The funny thing is you do home y and x and then qgl which of course doesn't work z homen

BOAndy1985, 2022-11-19:
khumarahn
variable_dock_on_zhome: false

khumarahn, 2022-11-19:
> khumarahn variable_dock_on_zhome: false

yeap, figured that out too. But it is still a bug

BOAndy1985, 2022-11-20:
What other error did you find?

khumarahn, 2022-11-20:
Just the original one

jlas1, 2022-11-21:
> Also I wonder what `variable_bypass_probe_docking` is?..

It's to allow an easier migration to Voron TAP probe

jlas1, 2022-11-21:
> Homen in z is no longer possible with the new version of macro. The error says to home x and y first before going to the dock

can you enable debug on klicky variables and paste the console here?

jlas1, 2022-11-21:
I think i found the bug, can you check the last version of the macros.

jlas1, 2022-11-25:
As it should be correct, i'm closing this issue, if it happens again, it can be reopened.

closed 2022-11-25 by jlas1

khumarahn, 2022-11-25:
Thanks a lot! Looks like it worked, will watch out