jlas1/Klicky-Probe #166: Bed mesh calibrate and probe calibration macro

Kamilleqq, 2022-11-20:
I have custom probe with electro deploying pin and these two commands doesn't work at all. 
They don't deploy pin, i have special code to deploy and measure height with this probe. The G28 work properly with custom macro of G28 but i don't know how i can customize the bed mesh calibrate and probe calibration macro. Can you help me with this?

jlas1, 2022-11-20:
check this for probe_calibrate:
https://github.com/jlas1/Klicky-Probe/blob/ea2cc64e5be37d56d4b4a01e080d8944aca75b0b/Klipper_macros/klicky-macros.cfg#L466

and this for probe:
https://github.com/jlas1/Klicky-Probe/blob/ea2cc64e5be37d56d4b4a01e080d8944aca75b0b/Klipper_macros/klicky-bed-mesh-calibrate.cfg#L14

It's pretty simple, you do a rename on the original macro, do what you want, then call the original renamed macro and do something again

jlas1, 2022-11-25:
closing this, if you need reopen please.

closed 2022-11-25 by jlas1