jlas1/Klicky-Probe #170: Trying to set up probe as Z endstop. Z homing sequence does not pick up probe.

cullinaire, 2022-11-28:
I got the klicky installation finished and working properly with the normal Z endstop per stock Voron. Next I wanted to use the probe as Z endstop so I followed the linked documentation. Then came the problems: at Z home, instead of attaching the probe, the printhead would go to the dock and act like it's detaching the probe even though it never had a probe in the first place. So I set variable_dock_on_zhome to false.
That fixed the docking issue, but now instead of attaching the probe, at Z home it just stops with error "Endstop z still triggered after retract". So I then inverted the polarity of the endstop pin. Now the probe is not picked up at Z home, and attempts to just home it anyways forcing me to emergency stop to prevent the nozzle from crashing.
The only way I have ever successfully done a Z home is by manually homing X and Y, then manually attaching the probe (which required me to reset polarity of the probe endstop pin (which now shows as normally triggered) and running the Z home.
[klicky-variables (1).zip](https://github.com/jlas1/Klicky-Probe/files/10101523/klicky-variables.1.zip)

cullinaire, 2022-11-29:
Error in printer.cfg: there was a space between "probe: z_virtual_endstop" in the stepper_z endstop pin section. Removed the space so it reads "probe:z_virtual_endstop" and all is fixed.

jlas1, 2022-11-29:
just would like to note that it's solved in the code now.

closed 2022-11-29 by jlas1