jlas1/Klicky-Probe #171: Gantry sagging while homing XY [Bug?]

HateFindingNames, 2022-12-20:
Printer: Voron 2.4

`variable_safe_z` and `variable_enable_z_hop` enables all four z-steppers before doing X and Y homing. Below line disables them again leading to considerable sagging of the rear side of the gantry during XY homing procedures ultimately leading to longer or even failed QGL.

https://github.com/jlas1/Klicky-Probe/blob/2bfb9bff0f7248ce583f53de4798b05bfb2be10c/Klipper_macros/klicky-macros.cfg#L665

Adding
`SET_STEPPER_ENABLE STEPPER=stepper_z ENABLE=1`
`SET_STEPPER_ENABLE STEPPER=stepper_z1 ENABLE=1`
`SET_STEPPER_ENABLE STEPPER=stepper_z2 ENABLE=1`
`SET_STEPPER_ENABLE STEPPER=stepper_z3 ENABLE=1`
right after line 665 fixes the issue (is there even a command to set the homing state w/o changing stepper states?).

adolfotregosa, 2022-12-26:
Why is there M84 in the first place ? Can someone explain, please?

HateFindingNames, 2022-12-26:
> Why is there M84 in the first place ? Can someone explain, please?

Your are right. For some reason I thought the homed states of all axes need to be cleared at this point but the whole section only gets executed if all axes are already in an unhomed state.

https://github.com/jlas1/Klicky-Probe/blob/2bfb9bff0f7248ce583f53de4798b05bfb2be10c/Klipper_macros/klicky-macros.cfg#L652

I will let my opening post unchanged for now. But as I don't see any reasy why at least Z needs to be de-energized I will change the title accordingly.

adolfotregosa, 2022-12-26:
Yeah, code forces Z to move upwards and then M84 would turn every stepper off.  I don't get the logic. That M84 seams out of place

darkpoetry, 2023-01-08:
This really caused me some terror on my first attempt to home my Voron V2 as the gantry plunged while it was homing x/y. Please remove it; it's likely to wreck any machine that doesn't have a gantry that will stay in a fixed position when unpowered.

okeightysix, 2023-01-09:
I tried MakersSource modified cfg for macros and it keeps motors engaged. You can try with some modifications I guess

https://github.com/MakersSource/KlipperBackup_SwitchwireOrange/blob/master/klicky-macros.cfg