jlas1/Klicky-Probe #173: Switchwire Drive Arm STL's are backwards

andypm1982, 2023-01-06:
the STL's i downloaded show the magnet holes are reversed on the arms. the magnet end up being on the wrong side of the arms.

How can i resolve this?

Thanks.

seansch, 2023-01-21:
I've been scratching my head about this too, the arms I printed don't match up with what is in the Switchwire installation guide pictures.

seansch, 2023-01-21:
I went ahead with the assembly with the printed parts from the SW STL folder and they are correct. 

But for reference what I was comparing to were the images here:
 https://github.com/jlas1/Klicky-Probe/blob/7a31e885b36d98702cce62d8181da80eb7d446b5/Printers/Voron/Switchwire/Photos/20220601_122838.png 
https://github.com/jlas1/Klicky-Probe/raw/7a31e885b36d98702cce62d8181da80eb7d446b5/Printers/Voron/Switchwire/Photos/20220601_123056.png
https://github.com/jlas1/Klicky-Probe/raw/7a31e885b36d98702cce62d8181da80eb7d446b5/Printers/Voron/Switchwire/Photos/back_servo.png
https://github.com/jlas1/Klicky-Probe/raw/7a31e885b36d98702cce62d8181da80eb7d446b5/Printers/Voron/Switchwire/Photos/IMG_20220606_1932032_2.jpg

These images show the magnets in different locations than the current printed parts.

jlas1, 2023-01-21:
Sorry about this, the parts were updated, to avoid having the magnets being pulled from the holes, but the images have not been updated.

closed 2023-01-21 by jlas1

andypm1982, 2023-01-21:
That’s fine and I get that but I’m finding the depth of the hole is to shallow now the magnets are reversed. The magnets even with triple stacked isn’t holding the arm in place. I don’t know whether that’s just when the servo isn’t powered or not. I need to test.

seansch, 2023-01-21:
I completed the assembly last night and didn't have any issues with the magnets holding the arms together, with or without the servo attached. Are you sure you have the arms attached correctly? The "small" holes should be facing each other, between the arms, ensuring that the polarity of the magnets is pulling the arms together.