jlas1/Klicky-Probe #175: FIRMWARE_RESTART fails when klicky-macros.cfg is enabled

calvinx408, 2023-01-08:
Hi, 
I got klicky working and it's awesome but I am un-able to do a firmware_restart so i have to do a node reboot. The only way firmware_restarts works is when I comment out klicky-macros.cfg from klicky-probe.cfg. I posted on discord but I didn't get any replies. Here are my klipper logs:
[klippy (2).log](https://github.com/jlas1/Klicky-Probe/files/10368646/klippy.2.log)
[klippy (3).log](https://github.com/jlas1/Klicky-Probe/files/10368644/klippy.3.log)

Thanks!

GadgetAngel, 2023-01-08:
> Hi, I got klicky working and it's awesome but I am un-able to do a firmware_restart so i have to do a node reboot. The only way firmware_restarts works is when I comment out klicky-macros.cfg from [...]

From your logs it looks like your problem is related to the communications with the mcu.  

I see the following error ```b'Got error -1 in read: (11)Resource temporarily unavailable'
TMC stepper_x failed to init: Timeout on wait for 'tmcuart_response' response```

I would be looking at the Klipper firmware version you are running on all your MCUs for your Klipper Printer and make sure all the mcu firmwares are all up to date

calvinx408, 2023-01-08:
I am running the latest for all. My klipper firmware is 
```
v0.11.0-57-gcbc17a7c
```

It's just weird that a node reboot fixes it all. Thanks again for your fast reply.

calvinx408, 2023-01-13:
I just updated the mcu so it's matching now. Still not able to FIRMWARE_RESTART but restart works again.

[image]

Here are the klipper logs:
[klippy (4).log](https://github.com/jlas1/Klicky-Probe/files/10414495/klippy.4.log)