jlas1/Klicky-Probe #187: Lots of inconsistencies with sensorless homing & Klicky

Print3DWorld, 2023-03-12:
Hello,

I am trying to find ways to stabilize the klicky probe with sensor less homing. Sensor less homing is not an exact science therefore it can alter the endstop position by a half mm sometimes. I have had only 2 successfully started prints since installing this klicky probe (before I had inductive stock probe), and usually it's related to probe failing to dock/pick-up or ill find the sensor attached sideways etc.

Anyway, what are some good ways to "stabilize" the klicky with sensor less homing? I will be glad to share my findings if I can tolerate them long enough, we actually use our printers to make money so I don't want to keep tinkering with this if it'll never be stable.

kyleisah, 2023-03-18:
@Print3DWorld Hey there! You are correct, sensorless homing is not an *exact* science, but is typically improved/good enough by setting a "homing" motor current so the homing is a little more consistent. You could simply go back to microswitches, or get your homing sequence nicely tuned with sensorless homing.

Print3DWorld, 2023-08-30:
> @Print3DWorld Hey there! You are correct, sensorless homing is not an _exact_ science, but is typically improved/good enough by setting a "homing" motor current so the homing is a little more [...]

Hello,

I managed to tune the system quite well and finally got it to work semi-consistently with sensorless homing. I had to ensure the SGT/sensitivity is very well tuned and to get repeatable results.

60mm/s homing speed
1 amp homing current and hold current (do not set them differently, found this out the hard way!)
75 sensitivty

closed 2023-08-30 by Print3DWorld