jlas1/Klicky-Probe #192: No adaptation for Mini Stealthburner

rochford77, 2023-04-05:
Looking into options for the voron 0.2 and the new mini-stealthburner.

cereal7802, 2023-05-01:
So I'm using the Mini Stealthburner + Orbiter 2.0 and found the original klicky front cowling mount fits on the miniSB. It doesn't perfectly contour to the shape of the printhead like it does on the Afterburner, but it does seem like it would work. It shouldn't take that much adjustment to make it fit the MiniSB better.

hubertron, 2023-05-21:
> So I'm using the Mini Stealthburner + Orbiter 2.0 and found the original klicky front cowling mount fits on the miniSB. It doesn't perfectly contour to the shape of the printhead like it does on the [...]

The screwholes don't line up for my Mini-Stealth. Did you not bend or stretch anything?

jlas1, 2023-06-11:
there is now a V0.2 version

closed 2023-06-11 by jlas1