jlas1/Klicky-Probe #198: Homing 2nd axis (sensorless) not working correct

DerJOSh01, 2023-06-09:
Hey there,
i hope that I'm right to reach out in here.

I'm using the BTT Octopus 1.1 with BTT TMC2209 in UART with sensorless homing.

While NOT included klicky_probe my homing is working fine. Axis 1 moves to the frame, axis 2 moves to the frame. Each axis retracts a bit after contact. I attach the probe manually, home Z - everything ok.

I write axis 1 & 2 because the issue happens regardless which axis homes first.

If I include klicky it works like the following. Axis 1 moves to the frame, retracts a bit. Axis 2 moves a tiny bit in homing direction and retracts. Terminal sayes axis homed.
This happens if I select home x/y only and if I select home all.

If I manually home axis 1 and after that axis 2 everything is working fine as well. After that i can home z properly - so the position of the dick is driven to correctly, it attaches, homes, docks and moves away. All like it should.

Before, I had an SKR Mini E3 V2 with physical Endstops and there everything works fine.

So could someone help me out with this?

Kind regards JOSh

jlas1, 2023-06-11:
what are your homing scripts. i believe you need to have a dwell time between homing X and Y, that's probably why it works manually

DerJOSh01, 2023-06-11:
Hi,
Thanks for the comment, but i need again a little push.
I don't have any homing scripts except the original one, I think. The only thing I do is pressing the buttons in fluidd main screen. I don't know there or what the button leads to.
Can you help me finding this or create a homing script?

Kind regards
JOSh

jlas1, 2023-06-15:
check this guide https://docs.vorondesign.com/community/howto/clee/sensorless_xy_homing.html, should help.
Klicky supports the _home_x and _home_y macros and uses them if they are found on the configuration.

DerJOSh01, 2023-06-15:
Thanks, you got it.
I followed the instructions and it mostly works.
The thing I had to figure out is, that I for what ever reason, don't have to include homing_oderride. There my probe always showed as triggered so the klicky taught the probe is attached and homes instantly. So i went for commenting out the z option in homing_override but it didn't work. At last I don't include it ant now it's working like it should!
Thanks again and kind regards.

closed 2023-06-15 by DerJOSh01