jlas1/Klicky-Probe #201: Klicky wont attach and detach correctly. Says already attached

Cipher43, 2023-06-12:
Hello.  Im trying to setup Klicky probe on my Ender 5 plus converted to Mercury 1.1 with a stealthburner.  I changed the few variables that it says I need to but when i test with the attach probe macro it says already attached.  When i test with dock probe macro it attaches the probe.   Also if I try and run a standard homing it will perform x and y correctly and then tries to probe without picking up the klicky probe.
Thanks

jlas1, 2023-06-15:
you may have a wiring issue, then the mcu thinks that the probe is always attached.
Can you try to do query_probe with:
probe at the dock (should get triggered)
probe mounted on the toolhead (should get open)
probe mounted on the toolhead and switch pressed (should get triggered)

Cipher43, 2023-06-19:
Ok so I did the query_probe and it says open when its on the dock.

Cipher43, 2023-06-19:
Ok thanks.  I had the wires from the probe backwards.  Now when I tell it to attach it moves sideways and then says   Error evaluating 'gcode_macro _SetProbeState:gcode': gcode.CommandError: Probe attach failed!

Cipher43, 2023-06-19:
Ok so i got that first part fixed after discovering a bad spot in a wire.  Now the probe says that its triggered when docked, open when attached to the toolhead, and triggered when the button is pressed while mounted.  When I use the macro attach probe, it moves only in the y direction and picks it up and then moves back in the y direction to drop the probe.  and give this error   Error evaluating 'gcode_macro _SetProbeState:gcode': gcode.CommandError: Probe attach failed!           if I manually move it to the attach position and query the probe it states its open as if it should be connected.  

For reference on this machine its an ender 5 plus converted to mercury one.1 with a stealthburner. For boards I am using a BTT skr mini e3 v2, BTT ebb sb2209, and u2c canbus on mainsail.  It wont let me post the actual printer.cfg or klickyvariables on here   so this is the best i can do.

Klickyvariables
# This macro was provided by discord user Garrettwp to whom i give my thanks for sharing it with me.
# I have tweaked it a lot.
# They are based on the great Annex magprobe dockable probe macros "#Originally developed by Mental,
# modified for better use on K-series printers by RyanG and Trails", kudos to them.
# That macro as since evolved into a klipper plugin that currently is pending inclusion in klipper, 
# more information here, https://github.com/Annex-Engineering/Quickdraw_Probe/tree/main/Klipper_Macros
# User richardjm revised the macro variables and added some functions, thanks a lot
# by standing on the shoulders of giants, lets see if we can see further
#
# the current home for this version is https://github.com/jlas1/Klicky-Probe 
# the 1000 values below is to give an error instead of doing something wrong, hopefully, this won't be used is a printer larger than 1 meter

[gcode_macro _User_Variables]
variable_verbose:             True    # Enable verbose output
variable_travel_speed:         400    # how fast all other travel moves will be performed when running these macros
variable_move_accel:          1000    # how fast should the toolhead accelerate when moving
variable_dock_speed:            300    # how fast should the toolhead move when docking the probe for the final movement
variable_release_speed:         300    # how fast should the toolhead move to release the hold of the magnets after docking
variable_z_drop_speed:          20    # how fast the z will lower when moving to the z location to clear the probe

variable_safe_z:         	    25    # Minimum Z for attach/dock and homing functions
# if true it will move the bed away from the nozzle when Z is not homed
variable_enable_z_hop:        True    # set this to false for beds that fall significantly under gravity (almost to Z max)

variable_max_bed_y:            355    # maximum Bed size avoids doing a probe_accuracy outside the bed
variable_max_bed_x:            355   # maximum Bed size avoids doing a probe_accuracy outside the bed

# if a separate Z endstop switch is in
# use, specify the coordinates of the switch here (Voron).
# Set to 0 to have the probe move to center of bed
variable_z_endstop_x:         0
variable_z_endstop_y:         0

#Check the documentation on klipper Dock/Undock configuration, these are dummy values
#dock location 
variable_docklocation_x:      12     # X Dock position
variable_docklocation_y:      343   # Y Dock position
variable_docklocation_z:      -128    # Z dock position (-128 for a gantry/frame mount)

#Dock move, final toolhead movement to release the probe on the dock
#it's a relative move
Variable_dockmove_x:             40
Variable_dockmove_y:             0
Variable_dockmove_z:             0
#Attach move. final toolhead movement to attach the probe on the mount
#it's a relative move
Variable_attachmove_x:           0
Variable_attachmove_y:           30
Variable_attachmove_z:           0

#Umbilical to help untangle the umbilical in difficult situations
variable_umbilical:          False    # should we untabgle the umbilical
variable_umbilical_x:           15    # X umbilical position
variable_umbilical_y:           15    # Y umbilical position

# location to park the toolhead
variable_park_toolhead:      False    # Enable toolhead parking
variable_parkposition_x:       125
variable_parkposition_y:       125
variable_parkposition_z:        30

variable_version:                1    # Helps users to update the necessary variables, do not update if the variables above are not updated

#Below this remark, you normally do not need to configure
#Attach move2
[... 497 lines cut ...]
# file named "firmware.bin" on an SD card and then restart the SKR
# mini E3 with that SD card.

# See docs/Config_Reference.md for a description of parameters.

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.110833, -0.006667, -0.100417
#*# 	-0.045000, -0.012500, -0.102500
#*# 	-0.084583, -0.137500, -0.237083
#*# tension = 0.2
#*# min_x = 5.0
#*# algo = bicubic
#*# y_count = 3
#*# mesh_y_pps = 2
#*# min_y = 5.0
#*# x_count = 3
#*# max_y = 300.0
#*# mesh_x_pps = 2
#*# max_x = 300.0

Cipher43, 2023-07-10:
https://youtu.be/6E9X7UoHcnE
https://youtu.be/LxS_Qxsc9KY

Ok so I am still stuck on this still.  I changed out the klicky probe I was using to see if that is the issue and it wasnt.  This one is a bought one instead of made and it has a led on it to show when its attached.  I took a video of the probe doing what I can get it to do.  The first video is mounting the probe by hand and running it through the probe query.  The second is actually trying to home the machine.  When it fails I manually move the machine 30 to the side so its in docking postion and then clicking z home.  It also fails to dock after it has finished.  
Thanks

closed 2023-07-18 by Cipher43