jlas1/Klicky-Probe #206: Parts list for voron 0.2r1 ?

bginvestor, 2023-07-03:
Hi guys,

I am not finding a parts list for the 0.2r1.  Its very confusing trying to piece the correct assembly.  Can you please organize a "common" zip assembly for 0.2r1?  It would be greatly appreciated.  Thank you.

bginvestor, 2023-07-03:
clarification.  ClickyNG for voron 0.2r1.

bginvestor, 2023-07-04:
I had to search and study several files, but I have the combination that works.  Still recommend organizing these files, it will deter some folks working on this project.   Pics and instructions need to be updated to 0.2r1

Johboh, 2023-08-27:
Can you share your findings :)?

I'm struggling to figure out what more to print for the toolhead mount when using https://github.com/jlas1/Klicky-Probe/blob/main/Printers/Voron/v0/v0_STL/V0.2r1_KlickyNG_MiniSB_Cowling.stl