jlas1/Klicky-Probe #211: klipper macros homing_override does not check current z height

miker1r6, 2023-07-15:
After some confusing crashes on my voron trident I found that if you home x or y with z already homed z will move to safe_z regardless of current z position I found this when testing and my slicer end script had a g28 x in it and it just rammed the nozzle into the print. 
the macro should check if the current position is greater than safe_z to decide if it is necessary to move the z.

Debug output:

10:56 am
_klicky_status_ready activating the LED STATUS_READY
10:56 am
probe: TRIGGERED
10:56 am
homing_override backing off X endstop, G0 X290.0 F12000
10:56 am
homing_override Homing X, G28 X0
10:56 am
homing_override moving toolhead to 25mm from 75.0mm in X homing seq
10:56 am
_klicky_status_homing activating the LED STATUS_HOMING
10:56 am
homing_override goint to home X
10:56 am
probe: TRIGGERED
10:56 am
moving to a safe Z distance
10:56 am
Homing X
10:56 am
G28 X

Nathan22211, 2023-07-20:
Same issue on my trident. Lost a print due to this

miker1r6, 2023-07-20:
> Same issue on my trident. Lost a print due to this

JosAr helped me solve it on the voron discord. 
I believe he was going to include the fix soon but in the meantime. 

You need to add this to each axis in the homing overide section of the klipper macros file.
[image]

jacksonliam, 2023-08-24:
This has also almost broke my printer on two occasions, after pressing cancel print and home (to get the nozzle away) it tried to first rapidly push the plastic back into the nozzle. 

@miker1r6  Thanks for the patch, I was just going to comment out those Z moves when homing x/y, that's a better fix.