jlas1/Klicky-Probe #246: is it possible for probe to stay undock on toolhead after homing all axis?

drumstick77, 2024-03-19:
is it possible for probe to stay undock on toolhead after homing all axis?

jlas1, 2024-03-19:
it is, you can do:
G28 X Y
ATTACH_PROBE_LOCK
G28 Z

Then when you want the probe to dock, do DOCK_PROBE_UNLOCK.

drumstick77, 2024-03-19:
thank you. Sorry, I probably should also say that I'm using the klicky macros. is it possible to adjust it rather than homing override? (I presume that's homing override that you've suggested?)

jlas1, 2024-03-20:
don't know if i understood your question correctly, those instructions are with klicky default homing override.
To be fair, these question appear quite often on initial setup, but dissapear as time goes by, because it's all automated, so a extra dock and undock goes unnotticed.

drumstick77, 2024-03-20:
Sorry for the confusion.
I'm still a bit confused as you can probably tell, I'm wondering if I should be updating the below code in the default homing override?
==============================================
it is, you can do:
G28 X Y
ATTACH_PROBE_LOCK
G28 Z
==========================

ebrune81, 2025-01-12:
I would like to improve the same process. Did you ever figure out how to incorporate the attach_probe_lock and dock_probe_unlock? I completely understand how the commands works, I just dont see where the commands fit in the flow.