jlas1/Klicky-Probe #248: Not attaching probe on G28 Z from mainsail after recent updates

53Aries, 2024-03-21:
After updating Klipper, mainsail, moonraker, everything a few days ago and updating my Klicky probe macros.cfg the printer no longer attempts to attach the probe when using the mainsail homing buttons. When clicking Home Z the toolhead moves to the center of the bed and error is raised "Endstop z still triggered after retract". No attempt is made to attach probe. Homing and probing works fine during print start macros. I don't know if something changed in mainsail or in the klicky macros, but something changed. My full printer config can be found at https://github.com/53Aries/SPOne.1

jlas1, 2024-03-21:
There was a klipper update the broke klicky macros, try to update to the last version and try again.

53Aries, 2024-03-22:
> There was a klipper update the broke klicky macros, try to update to the last version and try again.

I did update klicky along with all the other updates.

jlas1, 2024-03-22:
try to enable klicky debug and see what's it doing.
Also, if you are using klicky as a virtual Z endstop, check if it's properly configured, klicky is able to detect that and mount the probe before homing Z

AndyWo83, 2024-03-23:
I have almost the same problem
When the klicky probe ist attached, i have errors with ACCEL_TO_DECEL. 
This feature is deprecated and was replaced by MINIMUM_CRUISE_RATIO.
[image]

AndyWo83, 2024-03-23:
Here with klicky debug notifications...

14:44
Error on 'SET_VELOCITY_LIMIT ACCEL_TO_DECEL=': unable to parse
14:44
Error on 'SET_VELOCITY_LIMIT ACCEL_TO_DECEL=': unable to parse
14:44
Error on 'SET_VELOCITY_LIMIT ACCEL_TO_DECEL=': unable to parse
14:44
Error on 'SET_VELOCITY_LIMIT ACCEL_TO_DECEL=': unable to parse
14:44
Error on 'SET_VELOCITY_LIMIT ACCEL_TO_DECEL=': unable to parse
14:44
Attach_Probe moving from the dock to G0 X22 Y276 F4500
14:44
Attach_Probe moving to the dock with G0 X22 Y306 F3000
14:44
Attach_Probe moving to the dock with G0 X22 Y306 F3000
14:44
Attach_Probe moving near the dock with G0 X22 Y276 F15000
14:44
_klicky_status_busy activating the LED STATUS_BUSY
14:44
Attach_Probe going to attach probe
14:44
Attaching Probe
14:44
homing_override probe configured as a virtual Z endstop attaching probe
14:44
homing_override backing off Y endstop, G0 Y300.0 F15000
14:44
homing_override Homing Y, G28 Y0
14:44
homing_override backing off X endstop, G0 X300.0 F15000
14:44
homing_override Homing X, G28 X0
14:44
homing_override moving toolhead to 15mm from 0.0mm
14:44
homing_override no axis homed, setting position as X=Y=0 Z=0
14:44
_klicky_status_homing activating the LED STATUS_HOMING
14:44
homing_override goint to home all axes
14:44
probe: TRIGGERED
14:44
Homing Z
14:44
Homing Y
14:44
Homing X
14:44
X or Y not homed, forcing full G28
14:44
moving to a safe Z distance
14:44
No axis homed

AndyWo83, 2024-03-23:
I've updated your macro and now it works...

```
[gcode_macro _exit_point]
gcode:
    {% set function  = 'pre_' ~ params.FUNCTION %}
    {% set move  = params.MOVE|default(0) %}
    {% set speed = printer["gcode_macro _User_Variables"].travel_speed %}

    # mandatory to save the new safe position
    M400
    SET_VELOCITY_LIMIT ACCEL={printer.configfile.settings.printer.max_accel}
    SET_VELOCITY_LIMIT MINIMUM_CRUISE_RATIO={printer.configfile.settings.printer.minimum_cruise_ratio}
    RESTORE_GCODE_STATE NAME={function} MOVE={move} MOVE_SPEED={speed}
```

jlas1, 2024-03-23:
> I've updated your macro and now it works...
>
> ```
> [gcode_macro _exit_point]
> gcode:
>     {% set function  = 'pre_' ~ params.FUNCTION %}
>     {% set move  = params.MOVE|default(0) %}
>     {% set speed = [...]

Thanks for the reply, yours is a differrent issue than the one OP has, you can solve it by updating the macros, that issue was addressed on the lastest version.

bombsquad85, 2024-03-29:
> Thanks for the reply, yours is a differrent issue than the one OP has, you can solve it by updating the macros, that issue was addressed on the lastest version.

So I have updated my macros with the new versions on the github repository, however, I still can't home without homing x y then attach probe then Z. Did I miss a fix somewhere?

jlas1, 2024-03-29:
you can only attach the probe after doing a G28 X and Y, so not sure on whats the issue here.
If you are using klicky as a virtual endstop and it doesn't work if you only issue a G28, then please check the "virtual endstop" configuration, you may have a space after the virtual endstop config.

Can you paste your virtual endstop config?

bombsquad85, 2024-03-29:
> you can only attach the probe after doing a G28 X and Y, so not sure on whats the issue here. If you are using klicky as a virtual endstop and it doesn't work if you only issue a G28, then please [...]

I can post that this afternoon when I get home.

I'll check for spaces.