jlas1/Klicky-Probe #256: Klipper Macros and Sensorless Homing

seanmccully, 2024-04-24:
There is something about these MACROS that cause Sensorless Homing to consistently fail. It is always on the second AXIS.

 If Y is first than X will not HOME. 
If  X is first than Y will not HOME.

Especially with the TMC5160 drivers (at least on my printer) Sensorless Homing can be a bit finicky. But there is something in these MACROS, that feels like possibly a bug in Klipper. Doing a single axis homing for both X,Y than a G28 for all three axis(s) the second axis will still consistently fail. 

The above routine, at least on my printer will always clear any stickiness with sensorless homing on the TMC5160(s).

elSaxoo, 2024-04-30:
Hey, i had the same problem. For me using my own _HOME_X and _HOME_Y macros solved it for me. 

> XY Sensorless homing
If you are using sensorless homing, and have your own X and/or Y homing macros, you can use override the klicky macros behavior with your very own _HOME_X and _HOME_Y macros.
If they exist on your klipper configuration, klicky macro will use them instead of the default G28 commands.

I put the following into the printer.cfg and it works perfectly.

```
[gcode_macro _HOME_X]
gcode:
    # Always use consistent run_current on A/B steppers during sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.7 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}

    # Home
    G28 X
    # Move away
    G91
    G1 X10 F1200
    
    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}

[gcode_macro _HOME_Y]
gcode:
    # Set current for sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.7 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}

    # Home
    G28 Y
    # Move away
    G91
    G1 Y10 F1200

    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}

seanmccully, 2024-04-30:
I used those homing macros for a long time, and they just don't work consistently for me. In my current setup (48v, TMC5160, LDO2804AH)  I have found mostly by accident that the most consistent setting was not having any homing overrides. 

Setting the timers, changing the current just appears to add noise and inconsistencies in the results. Letting Klipper just do the sensorless homing appears to work 99.9% of the time. 

I need to investigate more into why,  Increasing SGT is not the answer, because while sometimes it will clear a motor not homing, other times it will be too high and skip steps.

janderholm, 2025-04-05:
I also had problems with the above macros. These work well for me:

```
[gcode_macro _HOME_X]
gcode:
    G28 X
    G90
    G0 X320 F6000
    M400 ;

[gcode_macro _HOME_Y]
gcode:
    G28 Y
    G90 
    G0 Y320 F6000
    M400
```

A guess is that the M400 is important to let everything run to a halt wheres the 1 second wait is just guesswork.

Be mindful of the 320 position in the macros. That's just a position that works well with my dock placement.